chipsalliance/rocket-chip #1689: Optimisation generated VCD files' size

noureddine-as, 2018-11-05:
**Type of issue**: enhancement suggestion

**What is the current behavior?**
Using Verilated emulator (also called C or C++ Emulator) generates VCD files that have a very big size (~ GiBytes) even for very simple 

**Enhancement request:**
According to the following documentation of [Verilator](https://www.veripool.org/projects/verilator/wiki/Manual-verilator), The `--trace` option (used in `emulator/Makefile`) enables the emulator to produce VCD waveforms, which are very huge especially when having a significant payload to be run.
As far as I know, the solution to overcome this is to use `vcd2fst` command to convert from VCD to FST, but this is annoying, especially if we don't have enough space on disk since the VCD file is created anyway, moreover the conversion takes some time to execute.
There is an option `--trace-fst` which is available in new versions of Verilator. However I don't know how far is updating verilator's version in RocketChip possible? 

Anyway this is just a suggestion (a low priority one may be?), I don't have any experience with Verilator, but I may help if I have enough pointers to a solution.

Thank you.
Best Regards!

noureddine-as, 2018-11-05:
@seldridge , nice and it works perfectly, thank you for the help!

Best regards.

closed 2018-11-05 by noureddine-as

NazerkeT, 2020-06-28:
Hi, @noureddine-as, could you, please, point (if you remember) what exact command has worked in your case to generate fst dump? I am facing the same situation and running above and related commands ended up unsuccessful.

Best regards,

noureddine-as, 2021-03-16:
Hi @NazerkeT,
Sorry I didn't see your question until today, accidentally.
Anyway, we ended up using the following command:
```bash
<path to emulator> -v - +max-cycles=<max cycles> <target executable> | vcd2fst - <path to the resulting waveform>.fst
```

Best Regards,

NazerkeT, 2021-03-23:
> Hi @NazerkeT,
> Sorry I didn't see your question until today, accidentally.
> Anyway, we ended up using the following command:
> 
> ```shell
> <path to emulator> -v - +max-cycles=<max cycles> <target executable> | vcd2fst - <path to the resulting waveform>.fst
> ```
> 
> Best Regards,

Hi, @noureddine-as
Thanks for the response!