chipsalliance/rocket-chip #1722: Data access replay error when crossingType=NoCrossing

juliusbaxter, 2018-11-25:
I'm using a design based on the SiFive Freedom E310, and I'm seeing 2 cycles of latency between the Rocket core issuing a data access and it appearing on the tile's TileLink interface.

The subsystem's config had RocketCrossingParams with crossingType = SynchronousCrossing(). I switched to NoCrossing to see if this would reduce the latency, and if so by how much.

It did indeed reduce the latency by 1 cycle each way.

However I then noticed an issue where the DCache would issue a valid pulse, issue an s2_nack back to the rocket core, but then the reissued address would be incorrect.

[image]

What happened is an offset calculation has been repeated, corrupting the address. My disassembly looks like so:

```
  1c007e:       10050537                lui     a0,0x10050
  1c0082:       0b050513                addi    a0,a0,176 # 100500b0 <spi_load+0xfe9001e>
  1c0086:       f14025f3                csrr    a1,mhartid
  1c008a:       058a                    slli    a1,a1,0x2
  1c008c:       95aa                    add     a1,a1,a0
  1c008e:       4188                    lw      a0,0(a1)
  1c0090:       8502                    jr      a0
```

The issue is that the ```add a1,a1,a0``` occurs again, resulting in an incorrect address in a1.

I've often wondered why the rocket core seems to replay instead of stalling while the various hazards or buses clear. Anyway, does this look like a real bug, and not a configuration issue? Is rocket supposed to work with NoCrossing?

aswaterman, 2018-12-06:
I'm not able to replicate this.  I wrote some identical code, except with different absolute addresses, but with the same code alignment; and I made it so that the first time the `lw` executes, `a.ready` is false, so the nack occurs.  After the pipeline flush, the program resumes executing at the `lw`, as expected; the `add` instruction is not repeated.

juliusbaxter, 2018-12-09:
Thanks for looking into this.

Our design has moved on a little bit since I filed the issue and I haven't been able to recreate the failure.

We are using some custom instruction fetch logic, and I can't rule out the bug being there. I thought I checked it at the time but I didn't store the waveform and I have fixed bugs in the frontend since I filed this. So this is likely a false alarm, sorry about that.

However, I do have an observation about a new issue I now see with the NoCrossing configuration. In trying to recreate this scenario, I configured 4 rocket harts with crossingType=NoCrossing again, and re-run our software. I then noticed the software wasn't progressing and noticed a deadlock of sorts. There is a semaphore register that 3 cores are attempting to claim (atomic swap instruction) and one core trying to release it by writing zero. The system becomes deadlocked because the hart trying to clear the register happens to see a deasserted ready from the tilelink bus every time it attempts the write.

[image]

I notice when we do have SynchronousCrossing set, the valid out from the tile is held until ready is asserted and the transaction is acked by the bus.

[image]

I guess the only way to get this behaviour is with SynchronousCrossing? I presume it'd take changes in the dcache to change the behaviour with NoCrossing?

Thanks again for looking into this. If I do see anything like this in future I'll be sure to save the trace.

aswaterman, 2018-12-10:
Thanks for the update on the previous issue.

On the latter one,  there is an implicit requirement of one storage element between the D$ and the network.  I'll submit a PR with a workaround.

aswaterman, 2018-12-10:
Is `SynchronousCrossing(BufferParams.flow)` appropriate for your use case?  This will prevent the livelock issue you observed, but retain the lower latency.

If so, then I think the right thing to do is to add a requirement to the D$ to ban the `NoCrossing` a.k.a. `SynchronousCrossing(BufferParams.none)` case.  I suspect the D$ is not the only component that makes this assumption, but it is a start.  @hcook sound ok?

PR #1743 (2018-12-11): Provide a modicum of TL.A buffering in the NoCrossing case
This prevents livelock in a multi-master system with NoCrossing.  Resolves #1722
closed 2018-12-11 by aswaterman