chipsalliance/rocket-chip #1737: Dcache Signals

ferdous313, 2018-12-06:
Hi,
I am trying to understand the Dcache.scala code. i am new to chisel.
I am trying to figure out the signals generated for a cache miss in Dcache.
What I understand is that s2 is the request from the core which is 2 clock cycles prior and s1 is the request 1 cycle prior.
I was wondering at a given time, which signal is generated for a cache miss, is it s2_valid_cache_miss?
Which request should I focus in case of the current request?
Any suggestion will be appreciated!
Thanks!
Nilufar

aswaterman, 2018-12-10:
Look at `tl_out_a.valid` and work backwards from there; you're right that that `s2_valid_cached_miss` is necessary but not sufficient to instigate a cache refill.

ferdous313, 2019-01-11:
@aswaterman I am also a bit confused about the replay action. What I understand is after a cachemiss the request is sent to the Memory via the MSHRFILES. When MSHRFILE get the data the data array is written with meta_write and replay is triggered incase if that was a readrequest. But if the request was write then no replay would be triggered. Please clarify.