chipsalliance/rocket-chip #1749: How to enable interrput?

sjtu-sp, 2018-12-12:
I want to know how to enable an interrupt

I tried to pull up an interrupt when Rocc have finished an instruction, in scala files, I add the interrupt val in RoccCommandRouter like this:

`class RoccCommandRouter(opcodes: Seq[OpcodeSet])(implicit p: Parameters)
    extends CoreModule()(p) {
  val io = new Bundle {
    val in = Decoupled(new RoCCCommand).flip
    val out = Vec(opcodes.size, Decoupled(new RoCCCommand))
    val busy = Bool(OUTPUT)
    val interrupt = Bool(OUTPUT)
  }`

and in AccumulatorExample calss, I set that each time AccumulatorExample finished a doAccum instruction,  it set the rocc_interrupt to high.

I found the rocc_interrupt signal in vpd file, in CSR module like the picture:
[image]

I can see the rocc_interrupt can be high, however, the pc and mepc didn't change, and MIE register was always 0, which means the Rocket Core cannot enter my interrupt routine. 

so how can I enable interrput?~

KevinLT8888, 2019-01-21:
I think you should write some software to enable the PLIC while running. The interrupt is not enabled by default. Check the sdk files, there are templates.

sjtu-sp, 2019-02-20:
> I think you should write some software to enable the PLIC while running. The interrupt is not enabled by default. Check the sdk files, there are templates.

I check my code, and I think I didn't connect RoCC interrupt to PLIC, however I don't know what to do. I found I can set N external interrputs, and I can find them in RTL code:`LevelGateway LevelGateway_1 ( // @[Plic.scala 170:27:freechips.rocketchip.system.MyConfig.fir@71764.4]
    .clock(LevelGateway_1_clock),
    .reset(LevelGateway_1_reset),
    .io_interrupt(LevelGateway_1_io_interrupt),
    .io_plic_valid(LevelGateway_1_io_plic_valid),
    .io_plic_ready(LevelGateway_1_io_plic_ready),
    .io_plic_complete(LevelGateway_1_io_plic_complete)
  );`
but I don't know how to connect it to RoCC in scala code.

ShuyunJia, 2019-04-18:
You can refer to my template program to enable interrupts: 
`https://github.com/ShuyunJia/baremetal-plic.git`

michael-etzkorn, 2023-01-15:
- Enable external interrupts in `mie` (bit 11)
- Write the priority threshold for your interrupt source 
    -  The source number depends on what other modules you have hooked up (and what order you hooked them up in) for which context RoCC will take. 
    - The generated DTS IO Map can be used to determine the source number. 
- Write priority threshold to 0 for your context (most likely context 0)
- Enable the interrupt for that source number on the PLIC 
- https://github.com/riscv/riscv-plic-spec/blob/8315aa04119f9cb846fbfe6ff0af343c45266df6/riscv-plic.adoc#2-memory-map

closed 2023-01-15 by michael-etzkorn