chipsalliance/rocket-chip #1752: Issue with DIV result not being written back to register file

tzwaenn, 2018-12-13:
**Type of issue**: bug report

**Other information**
On a Rocket core, I encountered the issue that the although the core accepts the result of the div (div.io_resp_valid & div.io_resp_ready), the data picked for the register file update (rf_wdata) is actually the dmem data. How did the core reach that situation? There is a memory instruction in the writeback state in that cycle that caught a npc_misaligned in the memory stage - so it turns to an exception in WB, allowing the div to write back its result by giving the div.io_resp_ready, but since the npc_misaligned case does not cause the activation of io_dmem_s1_kill, the dcache is free to actually give a response for this memory instruction, in particular, in case of a cache hit, the one that is causing the mux for rf_write to pick the dcache output instead of the div one.
I am attaching a waveform from a Scala-generated RTL. Please ignore the "ILLEGAL" text in the line otherwise showing the instruction - this is a debug signal from out verification environment that currently does not take the full instruction set into account.
Of course, this is on the Rocket level, not a full RocketTile. So can the Frontend deliver a misaligned PC like in this case? In general, it can - I have a configuration where misa is writeable, so take a write to clear the 'c' bit in misa followed by setting bit 1 in tvec followed by an instruction causing an exception. This requests the misaligned PC from tvec at the Frontend which it happily delivers, causing a situation where a non-CFI instruction causes a npc misaligned which is actually a "current PC misaligned".

[image]

aswaterman, 2018-12-20:
I'm pretty sure something you're doing is violating an implicit property of one of the interfaces.  mem_npc_misaligned should only ever be high for CFIs.

tvec bit 1 is hardwired to 0, so taking an exception can't result in the PC being misaligned.

tzwaenn, 2018-12-20:
Well, the attached trace was produced on the RocketTile - it takes some writes to CSRs to disable compressed instructions and then get to the situation where an address ending in ee is requested (value at the cursor position). So it seems in principle, the frontent can deliver misaligned PCs (this was obtained with formal verification - other traces could likely be found as well, this was just the shortest one).
[image]
I currently do not have a trace where this occurs together with the DIV (getting a cache hit takes more cycles I guess). I just say that in the case that the frontend delivers a misaligned PC, external units like mem are not properly cancelled if requested which may cause issues like the DIV I reported. Form an architecture point of view, it may make sense to have the core also cancel the external requests in mem stage for misaligned - so in the case that is unexpected in the overall system (pc misaligned), the core "just" causes the npc_misaligned, but no other side effects...

tzwaenn, 2018-12-20:
I am not trying this on software - just on the RTL. But I can extract the sequence of data words provided by the frontend from the simulation trace...
So I see 30199573 followed (CSR update of 301) by 3a082073 (CSR update of 3a0) and 10200073 (CSR update of 102). The next one then is the exception one: 100073 (a break?). The initial register values in the register file were picked as 0. And the initial PC (as it is an input of the RocketTile) was picked as 7ffffff8 (so just 31 bits used, not 38 like in the other issue). The instruction hitting the exception requests 657bcfee as PC.
This is on the RocketTile, so the Frontend is in the model as well.

aswaterman, 2018-12-20:
Can you just send me the waveform? Talking about this over github isn't working.

closed 2019-01-04 by aswaterman

tzwaenn, 2019-01-09:
I can confirm that the PC no longer gets misaligned, solving the reported DIV issue.

commit 9110beaf (2018-12-20): Move tvec bit-zapping from D to Q

This addresses a simulation-pessimism problem.  The synthesis result
will be the same either way.  Before, in simulation, the value held in
these registers prior to the first write might not have had the
appropriate bits zapped, because of random initialization.

Resolves #1752
commit 59d407bd (2018-12-20): Move tvec bit-zapping from D to Q

This addresses a simulation-pessimism problem.  The synthesis result
will be the same either way.  Before, in simulation, the value held in
these registers prior to the first write might not have had the
appropriate bits zapped, because of random initialization.

Resolves #1752
commit b0dfa0e4 (2018-12-20): Move tvec bit-zapping from D to Q

This addresses a simulation-pessimism problem.  The synthesis result
will be the same either way.  Before, in simulation, the value held in
these registers prior to the first write might not have had the
appropriate bits zapped, because of random initialization.

Resolves #1752
commit e21266dc (2018-12-20): Move tvec bit-zapping from D to Q

This addresses a simulation-pessimism problem.  The synthesis result
will be the same either way.  Before, in simulation, the value held in
these registers prior to the first write might not have had the
appropriate bits zapped, because of random initialization.

Resolves #1752