chipsalliance/rocket-chip #1753: RoCC Memory interface get stucked

aignacio, 2018-12-13:
Hi guys, 
I'm building an accelerator using RoCC interface and what's happening is that after querying some instructions the memory interface does not release ready signal (req_ready == '0'), so my FSM will not proceed waiting for memory interface availability...What do you think that could cause such problem? The address of req. it's right and it's the same for each red rectangle also, I'm repeating five times the same processing task in the accelerator.
[image]
[image]

tampler, 2019-01-04:
The main reason for ROCC to hang is that your logic doesn't assert io.resp.valid signal.

Rocket blocks the main thread after issuing an accelerator instruction. Thus, you need to make sure that every command to your ROCC is properly responded.

From what I see in your picture, you make a request to the memory with io.dmem.req.val, which is NOT granted with io.dmem.resp.val, which in turn locks your FSM and prevents acc to return to the main thread.

If you want a more detailed elaboration, pls submit a minimal viable test case

Luck

aignacio, 2019-01-07:
Hi @tampler,
thanks for the answer so do you mean that I always must answer with io.resp.valid to any cmd before I started some memory request like io.dmem.req.val == 1?

tampler, 2019-01-07:
This is required for valid main thread operation. Your ROCC command is issued as a CPU command and it's blocking. That means your program stucks until you return from your acc. 

It's your designer responsibility to respond to the main thread and maintain data coherency between CPU and acc, so that your acc can process data and put is somewhere, before executing the next command.

I recommend you to read Yunsup Lee papers and PhD thesis on Hwacha. The microarchitecture becomes somewhat sophisticated, but he shows how to solve such problems.

Luck!
Boris

Phantom1003, 2021-09-29:
@aignacio Can you share your experience solving this problem? I met the same situation. I try to wait resp_valid before I rise req valid, it still stalls to wait for req ready.
[image]

aignacio, 2021-09-29:
hey @Phantom1003, IIRC the reason behind is that you need to answer the CPU before working on your memory requests or do anything else, so if you have an acc coupled, first thing you do is to answer the req. with resp channel valid and then later you process it....

Phantom1003, 2021-09-29:
I use a queue to accept rocc commands, the io_req* and io_resp* in my screenshot is the rocc mem interface, not the cmd&resp interface.
In my mind, the rocc.resp interface should only be used when use `customx rs1, rs2, rd` format instruction.
In my design I only use `customx rs1, rs2` format, I found the resp tag becomes weird, I post on #2887 .  Which value did you use as your memory request tag?

tkf2019, 2023-04-30:
> @aignacio Can you share your experience solving this problem? I met the same situation. I try to wait resp_valid before I rise req valid, it still stalls to wait for req ready. [image]

I met the same problem. So can you show how you solved it? Thanks.

aignacio, 2023-05-01:
Hey @tkf2019, I do not have this code anymore also it has been over 5 years so cache was flushed unfortunately =/