chipsalliance/rocket-chip #1757: JAL and JALR store different return PC

tzwaenn, 2018-12-17:
**Type of issue**: bug report

**Other information**
In a 64-bit configuration with 40 bit addresses, sign extension is done when storing the return PC for JAL/JALR. However, there is a corner case where this is done differently for both instructions. At PC 0x7ffffffffc, JALR stores 0xffffff8000000000 as return PC whereas JAL stores 0x0000008000000000. Not sure if the bits above 39 are crucial in this case, but this at least looks weird...

aswaterman, 2018-12-17:
Thank you for the report.  In a loose sense, it doesn't really matter, since a program that attempts to return to 0x7ffffffffc + 4 should fail.  But the technically correct thing to do is to write rd = 0x08000000000.  (In other words, the definition of JAL[R] does not depend on the active virtual memory mode.)

aswaterman, 2018-12-17:
Can you provide instructions for replicating this behavior?  I can't make it happen.

tzwaenn, 2018-12-18:
On the RocketCore alone, it is trivial - just let the first PC be 0x7ffffffffc without any exception flags. On the RocketTile, the challenge is to get that PC without any exception flags - this at least cannot happen in 60 cycles from reset. So the interplay of all components in the RocketTile at least makes this very hard to trigger...

tzwaenn, 2018-12-18:
I am looking at a generated Verilog - it has 40 bit signals for PC and dmem address and should be default config (I did not generate the Verilog myself). And as the PC is a plain input in the RocketCore, I can "see" any value at the moment, in particular bit 39 set... Of coruse, this does not mean that the Frontend can produce a PC with bit 39 set without tagging it with some exception bit...

aswaterman, 2018-12-18:
mem_reg_pc should be more than 40 bits wide if 40-bit physical addresses are supported. I think if it’s 40 bits, then only 38-bit PAs are supported. If that’s the case, we have an explanation.

tzwaenn, 2018-12-18:
Thanks for your patience... All signals io_imem_req_bits_pc, io_imem_resp_bits_pc, mem_reg_pc are 40 bits.

aswaterman, 2018-12-18:
Yeah, that's the problem then... the core is only configured to support physical addresses below 2^38.

FWIW, we don't currently support physical addresses bigger than (virtual address bits - 1).  If you disable virtual memory, this constraint obviously disappears.

closed 2018-12-18 by aswaterman