chipsalliance/rocket-chip #1764: How add local interrupts to rocket chip

colin4124, 2018-12-24:
**Type of issue**: feature request

**Impact**:  API addition (no impact on existing code) 

**Development Phase**: proposal

**Other information**
 

I found local interrupts in [SiFive E31 Core Complex Manual](https://sifive.cdn.prismic.io/sifive%2Fb06a2d11-19ea-44ec-bf53-3e4c497c7997_sifive-e31-manual-v2p0.pdf) . I want to add this feature to my custom Rocket Chip too, but I can't found it in freedom e300artydevkit Config.

I found that Rocket Chip disable Local Interrupts as below:

1.  As the comment say `(they are connected to tile.intInwardNode in a seperate trait) ` , but I can't find this trait.

https://github.com/freechipsproject/rocket-chip/blob/7990d8ba1e14b386890a10b69e7288d026bc18fa/src/main/scala/subsystem/HasTiles.scala#L88-L130 

2.  `trait HasRocketTiles` connect interrupts without local interrupts

https://github.com/freechipsproject/rocket-chip/blob/4f0ca76fa52b0191ccb0a90194cbcf0b3d60dac0/src/main/scala/subsystem/RocketSubsystem.scala#L49

I want  to add external interrupts to Local Interrupts for real-time demand.  How do I implement that ?

There is a parameter `nLocalInterrupts` in `trait CoreParams`

https://github.com/freechipsproject/rocket-chip/blob/7990d8ba1e14b386890a10b69e7288d026bc18fa/src/main/scala/tile/Core.scala#L14-L29

I can change the numbers in this way:

```scala
class WithNLocalInterrupts(nLInt: Int) extends Config ((site, here, up) => {
  case RocketTilesKey => up(RocketTilesKey, site) map { r =>
    r.copy(core = r.core.copy(nLocalInterrupts = nLint))
  }
})
```

But I don't know how to connect external interrupts source to `RocketSubsystem` .

borancar, 2018-12-26:
I think part of this question would be better suited to [SiFive forums](https://forums.sifive.com/). I can share with you how I got them to work, with a disclaimer that I'm possibly doing it wrong. I've also ommited imports and defining the `NLocalExtInterrupts`, for brevity's sake. Here goes:

1. Taking inspiration from [InterruptBus.scala](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/subsystem/InterruptBus.scala#L34), you can create your own  `HasLocalExtInterrupts` trait  (note that `this` needs to be RocketSystem, since you will want `tiles` which represents CPUs):
   ```scala
   trait HasLocalExtInterrupts { this: RocketSubsystem =>
     private val device = new Device with DeviceInterrupts {
       def describe(resources: ResourceBindings): Description = {
         Description("local-external-interrupts", describeInterrupts(resources))
       }
     }
   
     val nLocalExtInterrupts = p(NLocalExtInterrupts)
     val localExtInterrupts = IntSourceNode(IntSourcePortSimple(num = nLocalExtInterrupts, resources = device.int))

     if (nLocalExtInterrupts > 0) {
       val asyncXing = LazyModule(new IntXing(3))
       asyncXing.intnode := localExtInterrupts
       tiles(0).crossIntIn() := asyncXing.intnode
     }
   }
   ```
   
   You need to add this trait to your [E300ArtyDevKitSystem](https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/System.scala#L24).

2. If you just want to expose them to the external world, you can create another trait for the Imp side, the sole purpose being creating the IO port and hooking it up to the :

   ```scala
   trait HasLocalExtInterruptsImp extends LazyModuleImp {
     val outer: HasLocalExtInterrupts
     val interrupts = IO(UInt(INPUT, width = outer.nLocalExtInterrupts))
   
     outer.localExtInterrupts.out.map(_._1).flatten.zipWithIndex.foreach { case(o, i) => o := interrupts(i) }
   }
   ```
   
   You need to add this trait to your [E300ArtyDevKitSystemModule](https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/System.scala#L37).

You can now connect your interrupts inside the [E300ArtyDevKitFPGAChip](https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/FPGAChip.scala).

borancar, 2018-12-26:
If, alternatively, you want to use your GPIOs as your interrupt sources, you will need to do more changes:
1. Remove the `HasPeripheryGPIO` and `HasPeripheryGPIOModuleImp` traits from [E300ArtyDevKitSystem](https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/System.scala#L31) and [E300ArtyDevKitSystemModule](https://github.com/sifive/freedom/blob/master/src/main/scala/everywhere/e300artydevkit/System.scala#L37), respectively.
2. Instantiate the GPIOs manually inside the E300ArtyDevKitSystem, similar to [unleashed/DevKitFPGADesign.scala](https://github.com/sifive/freedom/blob/master/src/main/scala/unleashed/DevKitFPGADesign.scala#L115), e.g.:
   ```scala
     val gpios = p(PeripheryGPIOKey).map { ps =>
       GPIO.attach(GPIOAttachParams(ps, pbus, ibus.fromAsync))
     }
   
     val gpioNodes = gpios.map { g =>
       g.ioNode.makeSink
     }
   ```

Now here's where it get really hacky, at least for me:
- I needed to add a port to TLGPIO manually as I couldn't use [interrupts](https://github.com/sifive/sifive-blocks/blob/a0da03f5a500917e71002a4b427893a49decc775/src/main/scala/devices/gpio/GPIO.scala#L191) from it inside E300ArtyDevKitSystemModule. Chisel3 kept giving me _T_xxx not defined.
- Instead of using `HasLocalExtInterruptsImp`, I manually connected them inside `E300ArtyDevKitSystemModule`:
   ```scala
     outer.localExtInterrupts.out.map(_._1).flatten.zipWithIndex.foreach {
       case(o, 0) => o := outer.gpios(0).module.lip(15)
       case(o, 1) => o := outer.gpios(0).module.lip(30)
       case(o, 2) => o := outer.gpios(0).module.lip(31)
     }
   ```

colin4124, 2018-12-27:
@borancar Thanks very much, I will have try.

terpstra, 2018-12-31:
To fan-out, use an IntXbar.

colin4124, 2019-01-02:
@borancar Thanks again, it works.

closed 2019-01-02 by colin4124