chipsalliance/rocket-chip #1778: AXI4 MEM port: wrong write address timing

marcinhl, 2019-01-15:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
The core issues a write on the MEM AXI4 interface with wrong timing of the awaddr. The right address appears one cycle late after awvalid, after the access has already been accepted with incorrect awaddress=0). This leads to a lockup as no response to access @0 comes back.
This is shown at the attached wave screenshot: the red marker shows the write with incorrect timing, the white one the preceding successful one.

Sw executed when it happens consists of a loop writing to 0x4xxxxxxx (multiple words) then performing a Dcache flush using fence.i (see below). The write with incorrect awaddr timing is the result of cache flushing.

I use Rocket for FPGA / simulation prototyping. The details on the Rocket build used:
 * Git commit hash: 4f0ca76fa52b0191ccb0a90194cbcf0b3d60dac0 ("fixing diamond problem")
 * based on DefaultConfig with some modifications (see below)

Modifications:
 * Config(new WithNBigCores(1) ++ new WithJtagDTM ++ new BaseConfig)
 * BaseConfig has WithNoSlavePort instead of WithDefaultSlavePort
 * WithDefaultMemPort address range tweaked: base=0x40000000 size=0x40000000
 * HellaCache.scala: flushOnFenceI set to true (I need cache flushing)

I had an older Rocket build with the same modifications that works with the same sw (Git commit hash: 0161220b3e0c76bf2d815592ff0f2ec80c55fb14)

**Please tell us about your environment:**
    - version: 4f0ca76fa52b0191ccb0a90194cbcf0b3d60dac0 
    - OS: `Linux version 3.10.0-514.26.2.el7.x86_64 (builder@kbuilder.dev.centos.org) (gcc version 4.8.5 20150623 (Red Hat 4.8.5-11) (GCC) ) #1 SMP Tue Jul 4 15:04:05 UTC 2017`

[image]

marcinhl, 2019-01-17:
I uploaded waves for the whole CPU, with gtkwave conf file (when loaded, shows the incorrect write). Please pick it up from here:
https://sendto.ultrasoc.com/pickup.php?claimID=KdDhNt3gJ2YaR47Y&claimPasscode=5oS3irsahCCdeEwF
(it's big, aroung 800MB)

terpstra, 2019-03-19:
Have you already resolved this issue? If not, can you please resend this waveform? The link does not work.

terpstra, 2019-03-19:
Just FYI, this adapter has been heavily verified, so really what needs to be diagnosed is your software. There's basically a 0% chance that the AXI request from the CPU is wrong.

marcinhl, 2019-03-19:
No, I haven't resolved it.
I re-uploaded the waves (sendto deletes files after a while, that's why the previous one doesn't work):
https://sendto.ultrasoc.com/pickup.php?claimID=4WPFtP7SJKRzWwEZ&claimPasscode=qQ7dwHpH9jJATxNe
> There's basically a 0% chance that the AXI request from the CPU is wrong.
There does seem to be a mistimed awaddr. I don't think there's much I can do in sw to influence interface timings like this.
Anyway, please have a look at the waves. It would be good to understand it

terpstra, 2019-03-19:
Ok. We've tracked down the cause. Thanks for the waveform! You are correct that it is not the software.

However, it's also not a problem with the AXI timing, but an interaction between the L1D and the flushOnFenceI feature in the design point you selected. There were a number of changes made to improve fence.i since the commit you are using. We believe this problem should be fixed in master. Can you try it and see if the problem is resolved?

marcinhl, 2019-03-20:
Thanks for looking at this. Yes, I'll try a recent commit on master, in the next few days and let you know whether this problem disappeared.

marcinhl, 2019-03-22:
I've regenerated my CPU from a recent hash 01030a97d2e6d924725b94c8ca731d302c441f65.
I can confirm that it now appears to work (at least one of the apps that consistently locked up now consistently works).
Thanks!

terpstra, 2019-03-25:
Great! I'm glad this issue has already been resolved. If for some reason you need to use the older version of rocket-chip, it should also suffice to disable flushOnFenceI.

closed 2019-03-25 by terpstra