chipsalliance/rocket-chip #1811: mmio (ExtBus) size ignored

gsomlo, 2019-01-30:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**What is the current behavior?**
The MMIO/ExtBus size parameter specified here:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/subsystem/Configs.scala#L328
which is also reflected in the generated DTS map, appears to be overridden and hard coded to 4KB here:
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/subsystem/Ports.scala#L119

**What is the expected behavior?**
The size of the generated MMIO region matches the size of the MMIO port specified for the design being generated.

gsomlo, 2019-02-03:
Thanks, that explains it!
Would it make sense to add a comment to that effect, e.g.
```
diff --git a/src/main/scala/subsystem/Ports.scala b/src/main/scala/subsystem/Ports.scala
index fc500bb7..3dedafb0 100644
--- a/src/main/scala/subsystem/Ports.scala
+++ b/src/main/scala/subsystem/Ports.scala
@@ -116,6 +116,7 @@ trait CanHaveMasterAXI4MMIOPortModuleImp extends LazyModuleImp {
 
   def connectSimAXIMMIO() {
     (mmio_axi4 zip outer.mmioAXI4Node.in) foreach { case (io, (_, edge)) =>
+      // test harness size capped to 4KB (ignoring p(ExtMem).get.master.size)
       val mmio_mem = LazyModule(new SimAXIMem(edge, size = 4096))
       Module(mmio_mem.module).io.axi4.head <> io
     }
```
I can make it a pull request if that's preferable, just let me know :)

hcook, 2019-02-09:
Sure, that would be a fine PR

closed 2019-02-12 by gsomlo