chipsalliance/rocket-chip #1826: What is the top module of rocket core ?

constatnt, 2019-02-07:
Hi, I have a question about rocket rtl code.

I only want to synthesis Rocket module (including rocket core and Rocc, not HITF)
Some people say the module "ExampleRocketSystem" is the top module of rocket core.

I think "ExampleRocketSystem" config includes unnecessary logic such as memaxi.

What is the top module of rocket core ?

CLappin, 2019-02-08:
Hi, 

The ExampleRocketSystem modules is what your looking for. 
The module TestHarness  instantiates the memaxi and the ExampleRocketSystem as designs under test (_dut) in the code.

A good test would be to remove those two modules and run just the ExampleRocketSystem as the top module.

Hope this helps
Ciaran

yswntht, 2019-07-31:
> Hi, I have a question about rocket rtl code.
>
> I only want to synthesis Rocket module (including rocket core and Rocc, not HITF)
> Some people say the module "ExampleRocketSystem" is the top module of [...]

look for `module Rocket`

CLappin, 2019-07-31:
module Rocket is the RISC-V HART itself as far as I know, which would only have the signals that connect to the TL2 interface. If you want to connect to external peripherals (GPIO, UART, DDR) then you will need the ExampleRocketSystem module. This module has the AXI4 interface or AHB, depending on how you have it set up.  The memaxi is used in the TestHarness when emulating the core as far as i know.

closed 2020-02-09 by mwachs5