chipsalliance/rocket-chip #1828: Make Linux run on top of the rocket-chip

nils1603, 2019-02-08:
Dear community,

First of all I want to say thank you for creating this project!

I am a total newcomer to everything related to RISC-V and I find it hard to get started.

So here is what I want to do. I want to have a fully functional Linux on my zedboard. I already looked into [fpga-zynq](https://github.com/ucb-bar/fpga-zynq), but I found that it was terribly outdated and I ended up fixing more than what I solved. Also I want to understand what is going on and not just execute some magic make scripts to make everything running.

I have experience with VHDL and Verilog. Personally I have a strong security background, but I am not a hardware engineer. I am very eager to learn, but right now I hope to be pointed in the right direction.

The problems I have encountered so far: With [fpga-zynq](https://github.com/ucb-bar/fpga-zynq) I ended up fixing more linking problems to make the scripts work to build u-boot, riscv-fesvr and everything for the ARM processor. Also I personally do not want to use the ARM processor. From my understanding it can simplify the process, but is not necessary. I understand that the [riscv-pk](https://github.com/riscv/riscv-pk) simplifies making programs run, but it should not be necessary?

I also stumbled across [riscv_vhdl](https://github.com/sergeykhbr/riscv_vhdl), which looks very interesting as well. 

From my general understanding. It should be possible to to generate my own rocket-core and make it run. So far I generated the chip, ran all the tests I could and generated the synthesizable Verilog file, which I imported to Vivado. I removed all the TestHarness stuff, so I was left with just the rocket-chip. I understand that I need an MMU, which should be built directly with default configuration, right?

What are the next steps? How can I connect everything. I probably need some DDR connected.

I also found [freedom](https://github.com/sifive/freedom). Is this worth looking more into for me? I probably can create my own fpag-shell for the Zedboard, but is this eventually going to help me achieve my objective to get Linux running?

It might be easier to get started by generating a chip and execute like a small program that transmits some stuff via UART? But how can I do that? 

Please point me in some directions. I am very eager to learn, but right now I feel very overwhelmed, with the variety of tools projects available. Probably someone has already done what I want, but google didn't help me much.

Thanks!

MyGithubAccount01, 2019-02-08:
I'm trying to do something similar--run Linux on rocket-chip. I've built Linux and it runs on spike. I've not had any luck translating that to rocket-chip. Have you looked at the notes on debugging with GDB? It might give you some clues.

ldelshad, 2019-02-08:
I'm not sure if my experience would help you or not. Although the prebuilt image from [fpga-zynq](https://github.com/ucb-bar/fpga-zynq) repository worked fine for me and I was able to bootup Linux on rocket-chip using zedboard, I needed to make some modifications to the Kernel and build the bbl from scratch. Unfortunately, the bbl generated from riscv-pk submodule of fpga-zynq didn't work for me. I wasn't even able to run that bbl with a dummy payload on the FPGA while things were working fine on spike/qemu.
To generate a working bbl for my FPGA, I ended up using an older commit of [riscv-pk](https://github.com/riscv/riscv-pk/tree/6b501de8e8512abd969199c57c5531737e52a105), [Linux kernel](https://github.com/riscv/riscv-linux/tree/fe92d7905c6ea0ebeabeb725b8040754ede7c220) 4.15 and loosely followed the [Linux installation manual](https://github.com/riscv/riscv-tools/tree/56f7b8ca7a1e2f8684899708cdb6d19a06e11926#-the-linuxrisc-v-installation-manual).

gsomlo, 2019-02-10:
Are you all aware of the lowRISC project (https://www.lowrisc.org/, https://github.com/lowRISC/lowrisc-chip) ? While it's mainly targeted at the nexys4ddr (artix7-based) board, they have branches targetting the zedboard (zynq), arty, kintex, etc.
Runs Debian, could probably run Fedora's risc-v port with only small changes.
There is also a set of minimal patches to rocket, linux, and bbl to get lowRISC, rebased to the latest upstream versions of each respective project here: http://www.contrib.andrew.cmu.edu/~somlo/lowRISC/

rcywongaa, 2021-06-04:
Probably relevant: https://github.com/litex-hub/linux-on-litex-rocket