chipsalliance/rocket-chip #1851: How to remove the cache coherence feature of  L1 data cache

zhouqinghua, 2019-02-21:
Hi, 
Currently, Rocket's L1 data cache supports cache coherence. How can I configure it to be a simple one which do not need data coherence? I have not found the related chisel cfg parameters.
thanks.

kritikbhimani, 2019-02-22:
you need to use https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/subsystem/Configs.scala#L169 . an example usage of which can be found here https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/system/Configs.scala#L76

zhouqinghua, 2019-02-27:
Oh, i got it. it seems the current solution is just adding a converter in tile's master node.
acquire operation -> get and release operation -> put.
but the coherence management logic is still existing in L1 dcache which will increase the circuit area.
is there any method to remove the redundancy logic in L1 dcache?

thanks very much.

kritikbhimani, 2019-02-27:
there is no method to remove the redundant logic in L1 dcache

closed 2019-03-20 by zhouqinghua