chipsalliance/rocket-chip #1861: illegal opcodes sometimes replayed

tzwaenn, 2019-03-04:
**Type of issue**: bug report

**Impact**: no functional change

**Other information**
I noticed that some opcodes, though identified as illegal, still do their normal computation in the pipeline even if the register file is not updated (they request source operands they are not going to use and access units like the dcache even if they already know the result does not matter). As an example, a C.FLDSP when mstatus.FS==0 is decoded as illegal, but still causes a memory request to the dcache and if that does not have a successful handshake (!dmem_req_ready), the C.FLDSP is replayed. Other replay reasons may also apply for illegal opcodes. Once there are no more replays, the illegal exception is raised as expected - so the architecture state update of the illegal opcode is as expected, there may just be side effects of the replay on timing and the other units.

tzwaenn, 2019-10-28:
For interrupts, the instruction in decode is properly "removed" from the pipeline - so no stalling, no access to cache/muldiv and no replay. Doing something similar for fetch-side exceptions and illegal instructions would improve worst case cycle counts and remove potential opportunities for side channel attacks...

aswaterman, 2019-10-28:
Sometimes, it is necessary to replay an illegal instruction - e.g. if the
frontend requests the instruction might be replayed.

In general, I don’t think it will actually improve worst-case time bounds,
as any particular replay behavior could be manifested by choosing a
different, legal instruction to execute instead. (Otherwise it would imply
that we had replay paths that exist for no reason.)

tzwaenn, 2019-11-04:
Right, I wrongly referred to worst case cycle count.
I meant to refer to a concrete instruction sequence where a certain instruction that is not tagged as replay by the frontend, but as having either a fetch-side exception or being illegal is treated like a normal instruction at the beginning of the pipeline (potentially stalled in decode to wait for operands, generating dcache requests and then cleanly killing them in the subsequent cycle). Such an instruction that will not update the architecture state, anyway, can also incur replays from data cache handshake and from the WB-to-EX forwarding paths for LW in case of a cache miss, for example. These replays (not the ones tagged from the frontend) seem unnecessary as it is clear that this instruction will not execute normally - it is either itself throwing an excetion or some eariler instruction already in the pipeline is.