chipsalliance/rocket-chip #1868: Non-standard opcode 0x30500073 in rocket chip standard configuration

tzwaenn, 2019-03-07:
**Type of issue**: other enhancement

**Impact**: unknown

**Development Phase**: proposal

**Other information**
 
During verification of the rocket chip against the RISC-V ISA, we found one opcode that the rocket chip accepts (at least in M mode), but that is not documented in the RISC-V ISA (user level or privileged) or the debug spec. Some digging in Scala and here suggests this was added in November 2018 as a "CEASE" instruction.
Should'n this be documented better?
As this is a non-standard opcode, should it not influence the misa register to reflect the presence of an instruction outside the RISC-V spec?

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2019-03-16:
Thanks for pointing out that we should be setting misa.X bit.

Sorry that the documentation for the CEASE instruction is lacking.  Informally, it stops the hart from retiring any more instructions until reset, including even as a result of debug interrupts.

closed 2020-02-09 by mwachs5

sequencer, 2024-06-29:
@aswaterman Is there any plan from the RVI perspective to standardize the `cease` instruction? There are three instructions in Rocket:
```
c.flush.d.l1 111111000000?????000000001110011
c.discard.d.l1 111111000010?????000000001110011"
cease 00110000010100000000000001110011
```
For `c.flush.d.l1` and `c.discard.d.l1`, they should be replaced by `Zicbom`, but for `cease`, it seems there is no ratified instruction for it.