chipsalliance/rocket-chip #1880: Is it possible to write a value to the address of memPort via mbus?

245950258, 2019-03-18:
I replace the axi interface with ahb interface. And I add a mem port which is connected to mbus.So,I have Defaultmemport,Defaultmmioport and extra mem port.
        The address map is shown below:
         Generated Address Map
	       0 -     1000 ARWX  debug-controller@0
	    3000 -     4000 ARWX  error-device@3000
	 2000000 -  2010000 ARW   clint@2000000
	 c000000 - 10000000 ARW   interrupt-controller@c000000
	10000000 - 20000000  RWXC memory@10000000
	20000000 - 30000000  RWX  mmio-port-ahb@20000000
	30000000 - 40000000  RWXC memory@30000000

         when I write a value to 0x34000000,the address is write ,but the hwrite signal is 0,and the hwdata is always 0.

hcook, 2019-03-18:
Memory assigned to the memory port is cacheable. If your design has caches, they will cache data upon accesses to those addresses. Writes will only become visible when the block is evicted from the cache.

245950258, 2019-03-19:
Thanks very much

closed 2019-03-19 by 245950258