chipsalliance/rocket-chip #1884: Memory address rearrangement questions

jasonlee1001, 2019-03-20:
I have a question about mem AXI port addressing.
The /subsystem/Configs.scala defines the mem and mmio port addresses.
The original setting of mem is:

 class WithDefaultMemPort extends Config((site, here, up) => {
  case ExtMem => Some(MemoryPortParams(MasterPortParams(
                      base = x"8000_0000",
                      size = x"1000_0000",
                      beatBytes = site(MemoryBusKey).beatBytes,
                      idBits = 4), 1)) 
})

I want to extend the size to x"2000_0000 , and also set the base = x"1000_0000"  but this cause the compilation to FAIL !

I tried to set size = x"3000_0000" , the compilation is pass , but the simulation is fail.
Generated Address Map
	       0 -     1000 ARWX  debug-controller@0
	    3000 -     4000 ARWX  error-device@3000
	   10000 -    20000  R X  rom@10000
	 2000000 -  2010000 ARW   clint@2000000
	 c000000 - 10000000 ARW   interrupt-controller@c000000
	10000000 - 20000000  RWXC memory@10000000  <-- why ?
	30000000 - 40000000  RWXC memory@10000000  <-- why ?
	60000000 - 70000000  RWX  mmio-port-axi4@60000000

My questions are:
  1. Are there any limitations about the mem address ? Can it begins from 0x1000_0000 ? Can the size must be less than 0x1000_0000 ?
  2. Why the memory address range be split into two address ranges ? 

Many thanks.

jasonlee1001, 2019-03-20:
I tried to set size = x"3000_0000" , the compilation is pass , but the simulation is fail.
--> I found the failure is because the 'SimAXIMem' module give the RRESP=DECERR to CPU when CPU issues the first burst transfer to acquire opcode which in SimAXIMem.
  But when the size = x"1000_0000" , the simulation is okay without DECERR.
Does anyone know why size matter ?

closed 2019-03-21 by jasonlee1001

245950258, 2019-03-22:
du you try to set size="0x1_0000".I use the ahb bus,and set the mem port size="0x1_0000".I failed to simulate.

jasonlee1001, 2019-03-24:
> du you try to set size="0x1_0000".I use the ahb bus,and set the mem port size="0x1_0000".I failed to simulate.

I never set that size because too small range can not be used in our system.
What is the begin address of your Mem port ?
AHB bus ? But the main bus of Rocket is AXI, do you add a axi-2-ahb wrapper by yourself, or Rocket can use AHB as its Top bus by modifying scala files ?

245950258, 2019-03-24:
You can modify scala files to use AHB just like AXI.
The Base address of Mem port is 0x1000_0000.I know the sizs="0x1_0000" is too small with a cache.So,I think it may be the problem of cache.