chipsalliance/rocket-chip #1889: option to disable verilator downloading

BobHanGood, 2019-03-21:
**Type of issue**: other enhancement

**Impact**: no functional change

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
cd vsim
make verilog
one behavior is downloading verilator and install it
**What is the expected behavior?**
think about offline users, they don't connect internet, i think we should add one option to switch the verilator downloading if or not. but not just downloading straightforward
**Please tell us about your environment:**

**What is the use case for changing the behavior?**

jasonlee1001, 2019-03-24:
By checking the 3 makefiles in /vsim, there are no 'verilator' in them.
According to the main page, run simulation in /vsim  you needs Synopsys vsim be installed.

> If you have access to VCS, you will be able to run assembly tests and benchmarks in simulation with the following commands (again assuming you have N cores on your host machine):
> 
> $ cd $ROCKETCHIP/vsim
> $ make -jN run CONFIG=DefaultFPGAConfig

BobHanGood, 2019-03-25:
> By checking the 3 makefiles in /vsim, there are no 'verilator' in them.
> According to the main page, run simulation in /vsim you need Synopsys VCS be installed.
> 
> > If you have access to VCS, you will be able to run assembly tests and benchmarks in simulation with the following commands (again assuming you have N cores on your host machine):
> > $ cd $ROCKETCHIP/vsim
> > $ make -jN run CONFIG=DefaultFPGAConfig

sorry, one typo, cd emulator