chipsalliance/rocket-chip #1904: Simulation with vsim hangs when executing WFI

hanoskoe, 2019-03-27:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 
Hi guys,
I am new to Rocket Chip and I could install the repo without problems. Then I tried to generate the files .out and .vcd in emulator and in vsim. What is strange for me is that I don't have any problems with emulator, but when generating the rv64ui-p-add.out file, when pc reaches the address 10054, it just hangs until it is stopped by the timeout (I once let it write until the end).

Is this a known issue or am I just doing something wrong? I am a freshman and don't understand what's going on. I didn't find any similar reported issue.

Thanks in advance

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
export RISCV= [pointing to riscv-tools]
cd RISCV
./build.sh
cd ../vsim
make output/rv64ui-p-add.out

**What is the current behavior?**
Here's the beginning of my rv64ui-p-add.out file:
`testing $random a15b114 seed 3324838479`
`C0:          0 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          1 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          2 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          3 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          4 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          5 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          6 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          7 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          8 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:          9 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         10 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         11 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         12 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         13 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         14 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         15 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         16 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         17 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         18 [0] pc=[00000000fa73d4f4] W[r 0=0000000000000000][0] R[r30=638608c6638608c5] R[r 8=d14466a2d14466a2] inst=[428f4e57] DASM(428f4e57)`
`C0:         19 [1] pc=[0000000000010040] W[r 0=0000000000000000][1] R[r 0=0000000000000000] R[r 1=88358f1188358f13] inst=[7c105073] DASM(7c105073)`
`C0:         20 [0] pc=[0000000000010040] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 1=88358f1188358f13] inst=[7c105073] DASM(7c105073)`
`C0:         21 [0] pc=[0000000000010040] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 1=0000000000000000] inst=[7c105073] DASM(7c105073)`
`C0:         22 [0] pc=[0000000000010040] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 1=0000000000000000] inst=[7c105073] DASM(7c105073)`
`C0:         23 [0] pc=[0000000000010040] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 1=0000000000000000] inst=[7c105073] DASM(7c105073)`
`C0:         24 [1] pc=[0000000000010044] W[r10=0000000000000000][1] R[r 0=0000000000000000] R[r20=88358f1188358f13] inst=[f1402573] DASM(f1402573)`
`C0:         25 [1] pc=[0000000000010048] W[r11=0000000000010048][1] R[r 0=0000000000000000] R[r 0=0000000000000000] inst=[00000597] DASM(00000597)`
`C0:         26 [1] pc=[000000000001004c] W[r11=0000000000010080][1] R[r11=0000000000010048] R[r24=88358f1188358f13] inst=[03858593] DASM(03858593)`
`C0:         27 [1] pc=[0000000000010050] W[r 0=3326496633264966][1] R[r 0=0000000000000000] R[r 4=88358f1188358f13] inst=[30405073] DASM(30405073)`
`C0:         28 [0] pc=[0000000000010050] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 4=88358f1188358f13] inst=[30405073] DASM(30405073)`
`C0:         29 [0] pc=[0000000000010050] W[r 0=0000000000000000][0] R[r 0=638608c6638608c7] R[r 4=88358f1188358f13] inst=[30405073] DASM(30405073)`
`C0:         30 [0] pc=[0000000000010050] W[r 0=0000000000000000][0] R[r 0=638608c6638608c7] R[r 4=88358f1188358f13] inst=[30405073] DASM(30405073)`
`C0:         31 [0] pc=[0000000000010050] W[r 0=0000000000000000][0] R[r 0=638608c6638608c7] R[r 4=88358f1188358f13] inst=[30405073] DASM(30405073)`
`C0:         32 [1] pc=[0000000000010054] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 5=88358f1188358f13] inst=[10500073] DASM(10500073)`
`C0:         33 [0] pc=[0000000000010054] W[r 0=0000000000000000][0] R[r 0=638608c6638608c7] R[r 5=88358f1188358f13] inst=[10500073] DASM(10500073)`
`C0:         33 [0] pc=[0000000000010054] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 5=88358f1188358f13] inst=[10500073] DASM(10500073)`
`C0:         33 [0] pc=[0000000000010054] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 5=88358f1188358f13] inst=[10500073] DASM(10500073)`
`C0:         33 [0] pc=[0000000000010054] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 5=88358f1188358f13] inst=[10500073] DASM(10500073)`
`C0:         33 [0] pc=[0000000000010054] W[r 0=0000000000000000][0] R[r 0=0000000000000000] R[r 5=88358f1188358f13] inst=[10500073] DASM(10500073)`
and then it stays forever like this (even the cycle count doesn't evolve)

**What is the expected behavior?**
I suppose at some point it should continue executing other instructions (as it was the case when I executed the same commands in the directory emulator)

**Please tell us about your environment:**

-rocket-chip commit `755e1718`
-OS:  `Linux i80pc127 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`
-VCS script version : `M-2017.03`

hanoskoe, 2019-04-03:
I also realized that when I type `make verilog` command in vsim directory, there is a "missing or empty reg/range property" warning, but I have the same warning for the emulator side and there the simulation works.
`[info] [0.004] Elaborating design...
Interrupt map (1 harts 2 interrupts):
  [1, 2] => dut

<stdout>: Warning (simple_bus_reg): Node /soc/external-interrupts missing or empty reg/ranges property
/dts-v1/;

/ {
	#address-cells = <1>;
	#size-cells = <1>;
	compatible = "freechips,rocketchip-unknown-dev";
	model = "freechips,rocketchip-unknown";
	L13: cpus {
		#address-cells = <1>;
		#size-cells = <0>;
		L6: cpu@0 {
			clock-frequency = <0>;
			compatible = "sifive,rocket0", "riscv";
			d-cache-block-size = <64>;
			d-cache-sets = <64>;
			d-cache-size = <4096>;
			device_type = "cpu";
			i-cache-block-size = <64>;
			i-cache-sets = <64>;
			i-cache-size = <4096>;
			next-level-cache = <&L8>;
			reg = <0x0>;
			riscv,isa = "rv64imac";
			status = "okay";
			timebase-frequency = <1000000>;
			L4: interrupt-controller {
				#interrupt-cells = <1>;
				compatible = "riscv,cpu-intc";
				interrupt-controller;
			};
		};
	};
	L8: memory@80000000 {
		device_type = "memory";
		reg = <0x80000000 0x10000000>;
	};
	L12: soc {
		#address-cells = <1>;
		#size-cells = <1>;
		compatible = "freechips,rocketchip-unknown-soc", "simple-bus";
		ranges;
		L2: clint@2000000 {
			compatible = "riscv,clint0";
			interrupts-extended = <&L4 3 &L4 7>;
			reg = <0x2000000 0x10000>;
			reg-names = "control";
		};
		L3: debug-controller@0 {
			compatible = "sifive,debug-013", "riscv,debug-013";
			interrupts-extended = <&L4 65535>;
			reg = <0x0 0x1000>;
			reg-names = "control";
		};
		L0: error-device@3000 {
			compatible = "sifive,error0";
			reg = <0x3000 0x1000>;
		};
		L7: external-interrupts {
			interrupt-parent = <&L1>;
			interrupts = <1 2>;
		};
		L1: interrupt-controller@c000000 {
			#interrupt-cells = <1>;
			compatible = "riscv,plic0";
			interrupt-controller;
			interrupts-extended = <&L4 11>;
			reg = <0xc000000 0x4000000>;
			reg-names = "control";
			riscv,max-priority = <3>;
			riscv,ndev = <2>;
		};
		L9: mmio-port-axi4@60000000 {
			#address-cells = <1>;
			#size-cells = <1>;
			compatible = "simple-bus";
			ranges = <0x60000000 0x60000000 0x20000000>;
		};
		L10: rom@10000 {
			compatible = "sifive,rom0";
			reg = <0x10000 0x10000>;
			reg-names = "mem";
		};
	};
};

Generated Address Map
	       0 -     1000 ARWX  debug-controller@0
	    3000 -     4000 ARWX  error-device@3000
	   10000 -    20000  R X  rom@10000
	 2000000 -  2010000 ARW   clint@2000000
	 c000000 - 10000000 ARW   interrupt-controller@c000000
	60000000 - 80000000  RWX  mmio-port-axi4@60000000
	80000000 - 90000000  RWXC memory@80000000`

Does anyone have a hint on that?

ShuyunJia, 2019-04-12:
I do not have vcs license at hand. It seems like you have problem executing the DPI-C function imported in SimDTM. The bootrom stops at WFI waiting for debug interface interrupt which is executed by a DPI-C function. You can look at the source code in `src/main/resources/csrc/SimDTM.cc` and `src/main/resources/vsrc/SimDTM.v`  

I suggest to check your vcs command line to confirm that dpi-c is enabled. The command line differs a little with different versions of vcs.

hanoskoe, 2019-05-06:
Thanks for your answer, but I didn't find how to enable that. I switched to an earlier version where I don't have those problems.

colinschmidt, 2019-05-06:
There was a recent bump to riscv-fesvr that in my experience has caused this hang at `wfi`. Did you install this update?

hanoskoe, 2019-05-07:
I am sorry I don't work on the updated version anymore and I don't have time to check this at the moment...

ShuyunJia, 2019-09-25:
I got the same issue after updating my repo and figured out why. 

I found out that the older version of fesvr is compiled into dynamic link file(.so) while the latest fesvr is compiled into static link file (.a). If one has built the previous version of fesvr, there would be a `libfesvr.so` in your `$RISCV/lib` directory, which cannot be overwritten by the new version. In the linking phase of your emulator, gcc would try to link libfesvr.so first, thus the older version of fesvr which is not compatible with your Debug module would be linked.

You can fix this by either of the two methods: 

1. remove the .so file in your $RISCV/lib directory
2. add `-static` flag into the 13 line of your `emulator/Makefile` so that gcc would try `libfesvr.a` file only.

@colinschmidt It looks like a common issue, maybe we should add this as tips in `README` or add the `-static` option in the `Makefile`. One thing to remind, the `-static` option would cause a increase in size of your emulator.

zhh124, 2020-04-09:
this seems caused due to rocketchip wait fesvr load the user program, and genereate a interrupt, the rocketchip jumpe to DRAM to run program. see bootrom/bootrom.S.

Inset below j instruction at _hang lable, to force rocket jump to _start:
_hang:
  j _start

abkarino, 2021-07-22:
I was using an old version of chipyard and it was working.
I made a new clone of chipyard (tried upgrade manually and didn't work), updated chisel3 and firrtl to latest release in both tools and rocketchip and built the sim but still stuck.
Using verilator.
```diff
diff --git a/build.sbt b/build.sbt
index d3e99cc9..57ca0c1c 100644
--- a/build.sbt
+++ b/build.sbt
@@ -72,7 +72,7 @@ def isolateAllTests(tests: Seq[TestDefinition]) = tests map { test =>
 // -- Rocket Chip --

 // This needs to stay in sync with the chisel3 and firrtl git submodules
-val chiselVersion = "3.4.1"
+val chiselVersion = "3.4.3"
 lazy val chiselRef = ProjectRef(workspaceDirectory / "chisel3", "chisel")
 lazy val chiselLib = "edu.berkeley.cs" %% "chisel3" % chiselVersion
 lazy val chiselLibDeps = (chiselRef / Keys.libraryDependencies)
@@ -81,7 +81,7 @@ lazy val chiselLibDeps = (chiselRef / Keys.libraryDependencies)
 //   keeping scalaVersion in sync with chisel3 to the minor version
 lazy val chiselPluginLib = "edu.berkeley.cs" % "chisel3-plugin" % chiselVersion cross CrossVersion.full

-val firrtlVersion = "1.4.1"
+val firrtlVersion = "1.4.3"
 lazy val firrtlRef = ProjectRef(workspaceDirectory / "firrtl", "firrtl")
 lazy val firrtlLib = "edu.berkeley.cs" %% "firrtl" % firrtlVersion
 val firrtlLibDeps = settingKey[Seq[sbt.librarymanagement.ModuleID]]("FIRRTL Library Dependencies sans antlr4")
diff --git a/generators/rocket-chip b/generators/rocket-chip
--- a/generators/rocket-chip
+++ b/generators/rocket-chip
@@ -1 +1 @@
-Subproject commit a7b016e46e22e4fdc013357051e30511f80df082
+Subproject commit a7b016e46e22e4fdc013357051e30511f80df082-dirty
diff --git a/project/plugins.sbt b/project/plugins.sbt
index 7e6f3aa8..fdc381a1 100644
--- a/project/plugins.sbt
+++ b/project/plugins.sbt
@@ -9,7 +9,7 @@ addSbtPlugin("org.scalastyle" %% "scalastyle-sbt-plugin" % "1.0.0")
 addSbtPlugin("com.eed3si9n" % "sbt-assembly" % "0.15.0")
 addSbtPlugin("com.simplytyped" % "sbt-antlr4" % "0.8.2")
 addSbtPlugin("com.github.gseitz" % "sbt-protobuf" % "0.6.3")
-addSbtPlugin("ch.epfl.scala" % "sbt-scalafix" % "0.9.21")
+addSbtPlugin("ch.epfl.scala" % "sbt-scalafix" % "0.9.25")
 addSbtPlugin("com.typesafe" % "sbt-mima-plugin" % "0.6.1")
 addSbtPlugin("com.eed3si9n" % "sbt-sriracha" % "0.1.0")
 addSbtPlugin("org.scalameta" % "sbt-mdoc" % "2.2.5" )
diff --git a/tools/chisel3 b/tools/chisel3
index 58d38f96..2554adfd 160000
--- a/tools/chisel3
+++ b/tools/chisel3
@@ -1 +1 @@
-Subproject commit 58d38f9620e7e91e4668266686484073c0ba7d2e
+Subproject commit 2554adfdae7933e7b0cf62ca71a6cb6b0c576f46
diff --git a/tools/firrtl b/tools/firrtl
index 7756f8f9..9c484be5 160000
--- a/tools/firrtl
+++ b/tools/firrtl
@@ -1 +1 @@
-Subproject commit 7756f8f9634b68a1375d2c2ca13abc5742234201
+Subproject commit 9c484be5ea2a0e7e90c07bff9bc8d4da8782a47a
```
Any clue on what might have caused this?