chipsalliance/rocket-chip #1921: Testbench Post Synthesis

vincenzociotoli, 2019-04-12:
Hello everyone,

What is the flow to do Testbench (for istance in VCS) after the generation of Verilog files in the folder "vsim" and after the synthesis in Synopsys Design Vision?

The 3 files generated are :  

example.TestHarness.DefaultConfig.harness.v
example.TestHarness.DefaultConfig.mems.v
example.TestHarness.DefaultConfig.top.v

I have synthesized the "Rocket Tile" in Design Vision with the last file "top.v" and now I want to do testbench after synthesis in VCS.
I guess that the testbench file is "harness.v" but I have read that this file serves only as an HTIF interface for the testbench files like "TestDriver.v" "SimSerial.v" "AsyncResetReg.v" etc.

Is this thought right?

Then, I have generated a Rocket Config that include a Custom Accelerator.
Do I have testbench files at the end that consider my accelerator?

Thank you all

colinschmidt, 2019-04-12:
I think this is a https://github.com/ucb-bar/project-template question, but I'll give you a quick answer here and then you should move further questions to that repo.
You should be synthesizing `top.v` with whatever module was set as `TOP` to synthesize the whole system, not just the tile.
Yes part of the test bench is in `harness.v` and the build system will include the extra test bench files you need like `TestDriver.v` etc.
If your custom accelerator has new top-level signals you would connect them in `TestHarness`  and they would then be included in `harness.v`

vincenzociotoli, 2019-04-12:
@colinschmidt thank you Colin! I don't know hot to delete this post or move it.

closed 2019-04-12 by colinschmidt