chipsalliance/rocket-chip #2033: Plz provide "libfesvr.so" file to simulate

sanjayvermapccs, 2019-07-12:
Hello Friends,
Due to some issue, I could not install fully rocket-tool. But I could successfully install partially. 
Can someone upload "libfesvr.so" file. this file is require to complete simulation.

It would be here:
$RISCV/lib/libfesvr.so

Thanks in advance.

CLappin, 2019-07-12:
Strange, I only installed this on a clean machine yesterday without issues. did you install the packages linked below before installing riscv-tools?

https://github.com/freechipsproject/rocket-tools/blob/master/README.md

sanjayvermapccs, 2019-07-12:
Thanks but I am following below link:
https://github.com/chipsalliance/rocket-chip

CLappin, 2019-07-12:
The packages I linked above are needed to build the rocket-tools.  Also, in the link there are instruction to specifically install riscv-tools for Mac OS X. I haven't tested it on  sierra , but I'd say this is a good place to start.  :) 

Thanks
Ciaran

wyatuestc, 2019-08-01:
Here is my solution. https://github.com/chipsalliance/rocket-chip/pull/2055

zhh124, 2020-04-09:
fesvr not contain in riscv-isa-sim project, but this project now not build a dynamic link library, only a static library will be build.
In this file riscv-isa-sim/fesvr/fesvr.mk.in, add below line then re-build, will generate libfesvr.so
fesvr_install_shared_lib = yes

FranDuqueAyachi, 2022-05-26:
okey, I added the line:
 fesvr_install_shared_lib = yes 
in the file fesvr.mk.in
but now, what do you mean with rebuild, i came from another project, the cva6 core and I have the same error when I try to simulate the core with a simple hello world program on questasim and I dont know where can i "rebuild" because it is already installed I there is no any script with a build of fesvr