chipsalliance/rocket-chip #2041: LIBTOOL issue while building rocket-tool

npandey2385, 2019-07-16:
./build.sh
Starting RISC-V Toolchain build process

libjaylink/Makefile.am:23: error: Libtool library used but 'LIBTOOL' is undefined
libjaylink/Makefile.am:23:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libjaylink/Makefile.am:23:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libjaylink/Makefile.am:23:   If 'LT_INIT' is in 'configure.ac', make sure
libjaylink/Makefile.am:23:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1

I tried to build riscv-fesvr, riscv-isa-sim, and riscv-pk only using the command:
./build-spike-only.sh

It works fine but it generate only  "libfesvr.a" library and I need "libfesvr.so" as well.

Could you pls help me here (How to fix LIBTOOL issue and how to generate ".so" file)

I am using the following version of gcc and other tools

OS:  Redhat 6
GCC:  6.3.0
libtool:  2.4.5
automake:  1.14.1
autoconf:  2.69

Thanks

npandey2385, 2019-07-22:
Hi,

This is little bit urgent , can someone pls help me on this.

CLappin, 2019-07-22:
Hi,

did you install the packages needed by riscv-tools?
A list of them is available in the read-me.
[Readme](https://github.com/freechipsproject/rocket-tools/blob/master/README.md)

Hope this helps and doesn't sound condescending.
Ciaran

wyatuestc, 2019-08-01:
https://github.com/chipsalliance/rocket-chip/pull/2055 here is my solution.

closed 2024-07-24 by michael-etzkorn