chipsalliance/rocket-chip #2042: Is there ECC support for rocket-chip in DefaultConfig?

clx7913, 2019-07-18:
In ECC.scala, in my opinion, there is a simple mechanism to select the type of ECC coding for the ICache and DCache that the generator finally will decide the ECC type by the string we assign to the "tagCode" and "dataCode".

In the "DefaultConfig" of Rocket, the "tagCode" and "dataCode" of both ICache and DCache are "None". So I think there is no ECC coding in the final verilog files. But I saw there are many ECC related codes in the final verilog files, such as "eccmask".

Then I don't know whether the ECC support effective in the "DefaultConfig" of Rocket.

`//selector of ECC type
object Code {
    def fromString(s: Option[String]): Code = fromString(s.getOrElse("none"))
    def fromString(s: String): Code = s.toLowerCase match {
        case "none" => new IdentityCode
        case "identity" => new IdentityCode
        case "parity" => new ParityCode
        case "sec" => new SECCode
        case "secded" => new SECDEDCode
        case _ => throw new IllegalArgumentException("Unknown ECC type")
    }
}

//the params of ICache
case class ICacheParams(
    nSets: Int = 64,
    nWays: Int = 4,
    rowBits: Int = 128,
    nTLBEntries: Int = 32,
    cacheIdBits: Int = 0,
    tagECC: Option[String] = None,
    dataECC: Option[String] = None,
    itimAddr: Option[BigInt] = None,
    prefetch: Boolean = false,
    blockBytes: Int = 64,
    latency: Int = 2,
    fetchBytes: Int = 4) extends L1CacheParams {
        def tagCode: Code = Code.fromString(tagECC)
        def dataCode: Code = Code.fromString(dataECC)
        def replacement = new RandomReplacement(nWays)
    }

//the params of ICache
case class DCacheParams(
    nSets: Int = 64,
    nWays: Int = 4,
    rowBits: Int = 64,
    nTLBEntries: Int = 32,
    tagECC: Option[String] = None,
    dataECC: Option[String] = None,
    dataECCBytes: Int = 1,
    nMSHRs: Int = 1,
    nSDQ: Int = 17,
    nRPQ: Int = 16,
    nMMIOs: Int = 1,
    blockBytes: Int = 64,
    acquireBeforeRelease: Boolean = false,
    pipelineWayMux: Boolean = false,
    clockGate: Boolean = false,
    scratch: Option[BigInt] = None) extends L1CacheParams {
    def tagCode: Code = Code.fromString(tagECC)
    def dataCode: Code = Code.fromString(dataECC)

    //the final verilog file
    assign data_io_req_bits_eccMask = dataArb_io_out_bits_eccMask;
    assign data_io_req_bits_way_en = dataArb_io_out_bits_way_en;
    assign dataArb_io_in_0_valid = pstore_drain_structural | _T_1732;
    assign dataArb_io_in_0_bits_addr = _T_1829[11:0];
    assign dataArb_io_in_0_bits_wdata = pstore2_valid ? pstore2_storegen_data : a_data;
    assign dataArb_io_in_0_bits_eccMask = {_T_1866,_T_1863};
}`

hcook, 2019-07-31:
> In the "DefaultConfig" of Rocket, the "tagCode" and "dataCode" of both ICache and DCache are "None". So I think there is no ECC coding in the final verilog files.

Correct.

clx7913, 2019-08-01:
OK, thank you

closed 2019-08-01 by clx7913