chipsalliance/rocket-chip #2043: DRET in debug mode outside of M mode illegal

tzwaenn, 2019-07-19:
**Type of issue**: bug report 

**Impact**: unknown

**Other information**
This is related to the previously reported and fixed #2022 abot DRET being legal outside of debug mode. The fix ensured DRET is illegal if outside of M mode or not in debug mode. So techinically, if you are in debug mode, but not in M mode, it is illegal.
And the trace I now see that makes this happen is essentially a debug interrput followed by SRET where SPP is U mode. So as this SRET executes normally in debug mode, we go to U mode, staying in debug. And the subsequent DRET is illegal.
I am not really sure if it should be possible to move outside of M mode if in debug mode - if so, the illegal computation of DRET is not according to spec. Otherwise, I assume M/S/URET should be illegal in debug mode to ensure we stay in M mode.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
debug interrupt - SRET to U mode - DRET

**Please tell us about your environment:**
The issue was found with the formal verification tool OneSpin 360 DV.

ernie-sifive, 2019-07-23:
Per the RISCV Debug Spec, section 4.1 (Debug):

_8. Almost all instructions that change the privilege level have undefined behavior. This includes
ecall, mret, sret, and uret. (To change the privilege level, the debugger can write prv in
dcsr). The only exception is ebreak. When that is executed in Debug Mode, it halts the
hart again but without updating dpc or dcsr._

GIven that, I believe the implementation is compliant with the spec.

tzwaenn, 2019-07-24:
I see. So executing an instruction with undefined behavior gets you in an undefined state, and in that state, even a well defined instruction like dret behaves not exactly according to spec in that state.
I still believe that tying DRET's illegal just to debug mode (and not debug and M mode) would be cleaner in terms of the spec (as that only talks of debug mode) and would also facilitate verification (dret always behaves as specified, even after an unspecified other RET).