chipsalliance/rocket-chip #2051: frontend's s0_speculative when usingCompressed

zhouqinghua, 2019-07-29:
Hi,
i have a question about the signal 's0_speculative ' in frontend module.
why this value is given to s0_speculative when usingCompressed is valid?
  // consider RVC fetches across blocks to be non-speculative if the first
  // part was non-speculative
  val s0_speculative =
    if (usingCompressed) s1_speculative || s2_valid && !s2_speculative || predicted_taken
    else Bool(true)

is there any special cases we need consider when usingCompressed?

thanks a lot.

aswaterman, 2019-07-31:
> to @aswaterman <https://github.com/aswaterman>.

For uncacheable memory, we only issue TL transactions for non-speculative
instruction fetches. Since 4-byte instructions can straddle cache-line
boundaries when usingCompressed, though, we need to issue a miss to the
next line when fetching the last two bytes in the current line. Otherwise,
we will make no progress, because the cache miss will never be satisfied.

The details of how the signal is set are due to the pipeline organization.

zhouqinghua, 2019-08-08:
yes, it should be that as your interpretation. s0_speculative should look the previous fetch action's speculative status. it seems s0_speculative should be set according s1_xxx signal(previous fetch). i don't understand the means of "s2_valid && !s2_speculative ".

my gussing  pseudo code:
if ((s1_valid == 1) && 
    (s1_speculative == 0) &&   //previous fetch is non-speculative
    (s1_pc[5:0] == 0b111110))  // previous fetch pc is non-4Byte aligned and cross cacheline(none rvc instruction)
    s0_speculative should force to 0 to give a opportunity to fetch the upper 2Btye regardless cacheable/uncacheable.

aswaterman, 2019-08-08:
I understand what you are getting at, but we have implicitly decided that it's OK to fetch the next cache line, provided the current cache line access is nonspeculative.  So we don't need the extra checks.  Note this is a platform decision, and it is consistent with the ISA spec.

What it means for users of the platform is that you should not place code immediately adjacent to a memory region that is marked executable and also has side effects.  We don't view this as a problematic constraint, because it's generally inadvisable to have memory regions that are both executable and side-effect-bearing.

closed 2019-08-22 by zhouqinghua