chipsalliance/rocket-chip #2052: ./build.sh failed at riscv-openocd

buinm, 2019-07-29:
**Type of issue**: bug report | feature request | other enhancement

I am trying to install rocket-tools but encountered this error when running ./build.sh. I have already installed all the required packages on https://github.com/freechipsproject/rocket-tools as well as setting the variable RISCV.
And this is blocking me from going further with using rocket-tools. 
Below is my errors:
Starting RISC-V Toolchain build process

Configuring project riscv-openocd
Building project riscv-openocd
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
ar: `u' modifier ignored since `D' is the default (see `U')
Makefile:4509: warning: overriding recipe for target 'check-recursive'
Makefile:3920: warning: ignoring old recipe for target 'check-recursive'
/bin/bash: line 9: /usr/local-linux/bin/makeinfo: No such file or directory
make[2]: *** [../doc/openocd.info] Error 127
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

Thanks,
Regards.
Minh

ShraddhaDevaiya, 2019-11-04:
facing the same! can anyone please help us to make the build successful?

HaFred, 2019-11-24:
https://github.com/chipsalliance/rocket-chip/issues/2137#issuecomment-539053809 #

AnushaGoel14, 2021-02-06:
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
In file included from ../src/flash/common.h:21,
                 from ../src/flash/nor/core.h:25,
                 from ../src/flash/nor/imp.h:22,
                 from ../src/flash/nor/jtagspi.c:22:
../src/flash/nor/jtagspi.c: In function ‘jtagspi_write_enable’:
../src/helper/log.h:130:2: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  130 |  log_printf_lf(LOG_LVL_ERROR, __FILE__, __LINE__, __func__, expr)
      |  ^~~~~~~~~~~~~
../src/flash/nor/jtagspi.c:255:11: note: ‘status’ was declared here
  255 |  uint32_t status;
      |           ^~~~~~
../src/flash/nor/jtagspi.c: In function ‘jtagspi_wait’:
../src/flash/nor/jtagspi.c:242:15: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  242 |   if ((status & SPIFLASH_BSY_BIT) == 0) {
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-011.c:18:
../src/target/riscv/riscv-011.c: In function ‘dbus_scan’:
../src/helper/binarybuffer.h:93:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   93 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-013.c:18:
../src/target/riscv/riscv-013.c: In function ‘dmi_scan.constprop’:
../src/helper/binarybuffer.h:55:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   55 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2897: src/openocd] Error 1
gmake[1]: *** [Makefile:4181: all-recursive] Error 1
gmake: *** [Makefile:1953: all] Error 2
anusha@anusha-VirtualBox:~/riscv-tools$ ./build.sh
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
In file included from ../src/flash/common.h:21,
                 from ../src/flash/nor/core.h:25,
                 from ../src/flash/nor/imp.h:22,
                 from ../src/flash/nor/jtagspi.c:22:
../src/flash/nor/jtagspi.c: In function ‘jtagspi_write_enable’:
../src/helper/log.h:130:2: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  130 |  log_printf_lf(LOG_LVL_ERROR, __FILE__, __LINE__, __func__, expr)
      |  ^~~~~~~~~~~~~
../src/flash/nor/jtagspi.c:255:11: note: ‘status’ was declared here
  255 |  uint32_t status;
      |           ^~~~~~
../src/flash/nor/jtagspi.c: In function ‘jtagspi_wait’:
../src/flash/nor/jtagspi.c:242:15: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  242 |   if ((status & SPIFLASH_BSY_BIT) == 0) {
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-011.c:18:
../src/target/riscv/riscv-011.c: In function ‘dbus_scan’:
../src/helper/binarybuffer.h:93:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   93 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-013.c:18:
../src/target/riscv/riscv-013.c: In function ‘dmi_scan.constprop’:
../src/helper/binarybuffer.h:55:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   55 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2897: src/openocd] Error 1
gmake[1]: *** [Makefile:4181: all-recursive] Error 1
gmake: *** [Makefile:1953: all] Error 2

same issue can anyone please help with it

ImagineMiracle-wxn, 2021-07-14:
@AnushaGoel14 HI! This blog can help you. Link: https://blog.csdn.net/qq_36393978/article/details/112509307