chipsalliance/rocket-chip #2054: Naively enabling debug system bus master causes crash

juliusbaxter, 2019-07-31:
I'm working with a rocket-chip clone which is from not too long ago (e44e22b) and I'd like to enable system bus access from the debug unit, so as to not trouble the harts when debugging the system.

If I naively set `hasBusMaster` like so in my config:

```
  case DebugModuleParams => DebugModuleParams(hasBusMaster = true, clockGate = true)
```

... then Chisel's elab stage crashes as below, due to some diplomacy issue it seems. I'm clueless as to what is missing beyond enabling the parameter. The example config used for regression `WithDebugSBASystem` offers no clues.

```
[info] [0.002] Elaborating design...
[error] (run-main-0) java.lang.reflect.InvocationTargetException
[error] java.lang.reflect.InvocationTargetException
[error]         at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error]         at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error]         at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities.$anonfun$elaborate$1(GeneratorUtils.scala:54)
[error]         at chisel3.core.Module$.do_apply(Module.scala:52)
[error]         at chisel3.Driver$.$anonfun$elaborate$1(Driver.scala:95)
[error]         at chisel3.internal.Builder$.$anonfun$build$2(Builder.scala:343)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:341)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at chisel3.internal.Builder$.build(Builder.scala:341)
[error]         at chisel3.Driver$.elaborate(Driver.scala:95)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities.elaborate(GeneratorUtils.scala:59)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities.elaborate$(GeneratorUtils.scala:50)
[error]         at freechips.rocketchip.system.Generator$.elaborate(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp.circuit(GeneratorUtils.scala:96)
[error]         at freechips.rocketchip.util.GeneratorApp.circuit$(GeneratorUtils.scala:96)
[error]         at freechips.rocketchip.system.Generator$.circuit$lzycompute(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.circuit(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp.generateFirrtl(GeneratorUtils.scala:102)
[error]         at freechips.rocketchip.util.GeneratorApp.generateFirrtl$(GeneratorUtils.scala:101)
[error]         at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:90)
[error]         at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:34)
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
[error]         at scala.App.$anonfun$main$1$adapted(App.scala:76)
[error]         at scala.collection.immutable.List.foreach(List.scala:389)
[error]         at scala.App.main(App.scala:76)
[error]         at scala.App.main$(App.scala:74)
[error]         at freechips.rocketchip.system.Generator$.main(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator.main(Generator.scala)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
[error]         at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
[error]         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
[error]         at java.lang.reflect.Method.invoke(Method.java:498)
[error]         at sbt.Run.invokeMain(Run.scala:93)
[error]         at sbt.Run.run0(Run.scala:87)
[error]         at sbt.Run.execute$1(Run.scala:65)
[error]         at sbt.Run.$anonfun$run$4(Run.scala:77)
[error]         at scala.runtime.java8.JFunction0$mcV$sp.apply(JFunction0$mcV$sp.java:12)
[error]         at sbt.util.InterfaceUtil$$anon$1.get(InterfaceUtil.scala:10)
[error]         at sbt.TrapExit$App.run(TrapExit.scala:252)
[error]         at java.lang.Thread.run(Thread.java:748)
[error] Caused by: java.lang.IndexOutOfBoundsException: 0
[error]         at scala.collection.LinearSeqOptimized.apply(LinearSeqOptimized.scala:63)
[error]         at scala.collection.LinearSeqOptimized.apply$(LinearSeqOptimized.scala:61)
[error]         at scala.collection.immutable.List.apply(List.scala:86)
[error]         at freechips.rocketchip.tilelink.TLXbar.$anonfun$node$5(Xbar.scala:46)
[error]         at freechips.rocketchip.diplomacy.MixedNexusNode.mapParamsU(Nodes.scala:518)
[error]         at freechips.rocketchip.diplomacy.MixedNode.liftedTree3$1(Nodes.scala:338)
[error]         at freechips.rocketchip.diplomacy.MixedNode.uiParams$lzycompute(Nodes.scala:335)
[error]         at freechips.rocketchip.diplomacy.MixedNode.uiParams(Nodes.scala:334)
[error]         at freechips.rocketchip.diplomacy.MixedNode.edgesIn$lzycompute(Nodes.scala:347)
[... 47 lines cut ...]
[error]         at freechips.rocketchip.util.HasGeneratorUtilities.$anonfun$elaborate$1(GeneratorUtils.scala:54)
[error]         at chisel3.core.Module$.do_apply(Module.scala:52)
[error]         at chisel3.Driver$.$anonfun$elaborate$1(Driver.scala:95)
[error]         at chisel3.internal.Builder$.$anonfun$build$2(Builder.scala:343)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at chisel3.internal.Builder$.$anonfun$build$1(Builder.scala:341)
[error]         at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
[error]         at chisel3.internal.Builder$.build(Builder.scala:341)
[error]         at chisel3.Driver$.elaborate(Driver.scala:95)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities.elaborate(GeneratorUtils.scala:59)
[error]         at freechips.rocketchip.util.HasGeneratorUtilities.elaborate$(GeneratorUtils.scala:50)
[error]         at freechips.rocketchip.system.Generator$.elaborate(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp.circuit(GeneratorUtils.scala:96)
[error]         at freechips.rocketchip.util.GeneratorApp.circuit$(GeneratorUtils.scala:96)
[error]         at freechips.rocketchip.system.Generator$.circuit$lzycompute(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.circuit(Generator.scala:12)
[error]         at freechips.rocketchip.util.GeneratorApp.generateFirrtl(GeneratorUtils.scala:102)
[error]         at freechips.rocketchip.util.GeneratorApp.generateFirrtl$(GeneratorUtils.scala:101)
[error]         at freechips.rocketchip.system.Generator$.generateFirrtl(Generator.scala:12)
[error]         at freechips.rocketchip.system.Generator$.delayedEndpoint$freechips$rocketchip$system$Generator$1(Generator.scala:90)
[error]         at freechips.rocketchip.system.Generator$delayedInit$body.apply(Generator.scala:12)
[error]         at scala.Function0.apply$mcV$sp(Function0.scala:34)
[error]         at scala.Function0.apply$mcV$sp$(Function0.scala:34)
[error]         at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
```

juliusbaxter, 2019-07-31:
Hmm something tells me I should try @hcook 's series of changes which went in a couple of weeks ago:

https://github.com/chipsalliance/rocket-chip/commits/master/src/main/scala/devices/debug

juliusbaxter, 2019-07-31:
Nope, merged with master and get the same error.

hcook, 2019-07-31:
My changes would affect to which internal crossbar that port of the debug unit attaches, but there is a default value which should preserve the original behavior (which I would expect to work in the generic `RocketSubsystem` example).

1. The type of error you are seeing (calling out in particular `Xbar.scala:46`) looks like your system has created a network graph in which the front bus xbar has the debug SBA attached as a client, but is itself connected to no managers. This is a surprising result given the suggested inclusion of [an optional connection between the front bus to the system bus](https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/subsystem/BusTopology.scala#L171), but perhaps you have rolled your own bus hierarchy and not included such a connection?

2. If you intentionally don't have a FrontBus and would like to continue to not have one, you could try `case ExportDebug => up(ExportDebug, site).copy(masterWhere = SBUS)` (this configurability requires my recent changes and was in fact the point of them).

(3. Alternatively you are somehow causing your diplomatic graph to be evaluated [perhaps by calling `.module` on a `LazyModule` instance] before you have finished constructing the whole graph, hence the missing manager connection. But let's explore the simpler explanations first.)

Let me know about 1 or see how 2 turns out and i will try to beef up the error message and example configuration.

juliusbaxter, 2019-08-01:
Thanks very much for your response @hcook, indeed the problem is that we don't have an fbus.

Adding `case ExportDebug => up(ExportDebug, site).copy(masterWhere = SBUS)` to our `Config.scala` gets us unstuck, thanks again!

Why no fbus: We don't have an fbus because when we try and use `HasHierarchicalBusTopology` elab breaks saying:

```
[error] Caused by: java.lang.IllegalArgumentException: requirement failed: memory_bus_xbar.node (A nexus node with parent 'memory_bus_xbar/mbus/sys' at  (MemoryBus.scala:56:32)) has 1 required inputs and no possible outputs
```

To get around that we just copy a portion of the trait:

```
 // The sbus masters the cbus
 sbus.crossToBus(cbus, NoCrossing)
 // The cbus masters the pbus
 cbus.crossToBus(pbus, NoCrossing)
```

We have no caches or any coherent  masters, nor a memory bus in our design. I'm slightly curious to know what is trying to attach to the mbus (which somehow survives without it in our usual configuration), but I won't lose sleep over it.

Again, appreciate the quick response!

closed 2019-08-01 by juliusbaxter