chipsalliance/rocket-chip #2084: Error-[FAE] File access error   File    "/home/hello/Install/rocket/rocket-tools-32bit/riscv-gnu-toolchain/lib/libfesvr.so"   cannot be opened for reading due to 'No such file or directory'.   Please fix this issue and compile again.  Makefrag:69: recipe for target 'simv-freechips.rocketchip.system-DefaultConfig' failed

ankun6, 2019-08-19:
Error-[FAE] File access error
  File 
  "/home/hello/Install/rocket/rocket-tools-32bit/riscv-gnu-toolchain/lib/libfesvr.so"
  cannot be opened for reading due to 'No such file or directory'.
  Please fix this issue and compile again.

Makefrag:69: recipe for target 'simv-freechips.rocketchip.system-DefaultConfig' failed

wyatuestc, 2019-08-28:
https://github.com/chipsalliance/rocket-chip/pull/2055

zhh124, 2020-04-09:
fesvr not contain in riscv-isa-sim project, but this project now not build a dynamic link library, only a static library will be build.
In this file **riscv-isa-sim/fesvr/fesvr.mk.in**, add below line then re-build, will generate libfesvr.so
**fesvr_install_shared_lib = yes**