chipsalliance/rocket-chip #2123: configurable PMA in machine mode

ShuyunJia, 2019-09-23:
**Type of issue**: feature request

**Impact**: API addition (no impact on existing code)

**Development Phase**: request

**Other information**
 
The S Mode is not always implemented in embedded SoCs. In current Rocket-chip, the PMA registers cannot be altered. Thus the memory attributes are not configurable by software. 

The multi-heterogeneous-core communication(e.g. IPC), however, is mostly realized via techniques such as virtio IO, which requires a piece of memory that can be configured uncacheable. In the current Rocket-chip, one has to configure the area as uncacheable during core generating and makes that area always uncacheable for all other applications.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
PMA registers cannot be altered by software

**What is the expected behavior?**
PMA registers can be configured by software

**Please tell us about your environment:**

**What is the use case for changing the behavior?**
Embedded IPC without MMU

aswaterman, 2019-10-11:
I understand the need, but we are not interested in adding writable PMAs yet.  One issue is that the format of these registers, which PMAs should be writable, etc. has not yet been architected and standardized.

For 64-bit systems at least, we have been planning to handle cacheability control with an address-space alias (i.e. the same memory would appear multiple times in the address space but with different PMAs).  This avoids the extra configuration state and also gives lower privilege modes controllability over cacheability of memory they can access.

I saw your proposed patch, and the overall approach makes sense to me, but I think for the time being, you will need to maintain it in your own fork.

ShuyunJia, 2019-10-12:
It seems a better way to use address alias for cacheability. It is clean and neat. Thanks so much for sharing. Issue closed.

closed 2019-10-12 by ShuyunJia

ShuyunJia, 2019-10-12:
And, if you need some feedback on how to format PMAs or other related issues, I'm glad to help :)

aswaterman, 2019-10-12:
Thanks for the offer :)