chipsalliance/rocket-chip #2137: Not able to setting up the RISCV enivronment

angi10, 2019-10-03:
I'm installing the RISCV environmet. I was launching these commands
$ git clone https://github.com/freechipsproject/rocket-tools
$ cd rocket-tools
$ git submodule update --init --recursive
$ export RISCV=/path/to/install/riscv/toolchain
$ export MAKEFLAGS="$MAKEFLAGS -jN" # Assuming you have N cores on your host system
$ ./build.sh

when i type ./build.sh, the system gives me some warnings and errors. I'll show you.

ar: `u' modifier ignored since `D' is the default (see `U')

/usr/bin/mkdir: /usr/bin/mkdir: impossibile creare la directory "/path"impossibile creare la directory "/path": Permesso negato: Permesso negato

Translation: Cannot create directory /path : permission denied

make[3]: *** [Makefile:2008: install-binPROGRAMS] Error 1
make[3]: *** Attesa per i processi non terminati....
make[3]: *** [Makefile:4130: install-dist_ulinkDATA] Error 1
make[2]: *** [Makefile:4489: install-am] Error 2
make[1]: *** [Makefile:4181: install-recursive] Error 1
make: *** [Makefile:4483: install] Error 2

Please someone help me, i've tried a lot of methods for changing the permissions, but i can't solve this issue

angi10, 2019-10-07:
Ok, now i have another type of error.
I have created a directory rocket-chip/toolchain, so i choose: 
`export RISCV=rocket-chip/toolchain`
now it appears to me, after `./build.sh`:
`configure: error: expected an absolute directory name for --prefix: rocket-chip/toolchain`

baltazarortiz, 2019-10-07:
Looks like you need to specify the full path to `rocket-chip` (i.e. `/home/...` or wherever else it lives)

angi10, 2019-10-08:
You're right, the problem was a wrong path that i used.
I think i've succesfully installed all.
Thanks everyone

closed 2019-10-08 by angi10

baltazarortiz, 2019-10-09:
No problem, glad to hear!

AnushaGoel14, 2021-02-06:
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
Makefile:5353: warning: overriding recipe for target 'check-recursive'
Makefile:4181: warning: ignoring old recipe for target 'check-recursive'
In file included from ../src/flash/common.h:21,
                 from ../src/flash/nor/core.h:25,
                 from ../src/flash/nor/imp.h:22,
                 from ../src/flash/nor/jtagspi.c:22:
../src/flash/nor/jtagspi.c: In function ‘jtagspi_write_enable’:
../src/helper/log.h:130:2: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  130 |  log_printf_lf(LOG_LVL_ERROR, __FILE__, __LINE__, __func__, expr)
      |  ^~~~~~~~~~~~~
../src/flash/nor/jtagspi.c:255:11: note: ‘status’ was declared here
  255 |  uint32_t status;
      |           ^~~~~~
../src/flash/nor/jtagspi.c: In function ‘jtagspi_wait’:
../src/flash/nor/jtagspi.c:242:15: warning: ‘status’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  242 |   if ((status & SPIFLASH_BSY_BIT) == 0) {
      |       ~~~~~~~~^~~~~~~~~~~~~~~~~~~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-011.c:18:
../src/target/riscv/riscv-011.c: In function ‘dbus_scan’:
../src/helper/binarybuffer.h:93:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   93 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
In file included from ../src/jtag/jtag.h:25,
                 from ../src/target/riscv/riscv-013.c:18:
../src/target/riscv/riscv-013.c: In function ‘dmi_scan.constprop’:
../src/helper/binarybuffer.h:55:19: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   55 |     buffer[i / 8] &= ~(1 << (i % 8));
      |                   ^~
/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/anusha/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2897: src/openocd] Error 1
gmake[1]: *** [Makefile:4181: all-recursive] Error 1
gmake: *** [Makefile:1953: all] Error 2

can anyone please help me with this