chipsalliance/rocket-chip #2151: why do I get an error when executing “make verilog”

AnouarNechi, 2019-10-16:
I was trying to begin with Rocket-chip and I was following the steps on the GitHub repository. Everything went fine until I tried to build the VCS simulator:
```
$ cd vsim
$ make
```
But I read in the issues that this tools is not free (have to pay for it ?!). So, I assumed that it is optional. and then I tried o generate FPGA- or VLSI-synthesizable Verilog:
```
$ cd vsim
$ make verilog
```
And I got the following error:
```
find: ‘/home/user/rocket-chip/api-config-sifive/design/craft/src/main/scala’: No such file or directory
make: Nothing to be done for 'verilog'.
```
I tried to look for the directory and it seemed that this is some kind of bug because that path does not exist in the repository. The closest thing to it is the following path  _**/home/user/rocket-chip/src/main/scala**_ where I have the following directories:
```
amba     
diplomacy              
groundtest  
jtag           
regmapper  
scie       
system  
tilelink  
util
devices  
diplomaticobjectmodel  
interrupts  
package.scala  
rocket     
subsystem  
tile    
unittest
```
Is this the right path ? if yes how can I correct this (which makefile should modify)

hyf6661669, 2019-10-19:
VCS is not a free simulator, you have to pay for it. You can use verilator (cd emulator) for simulation, which is opensource and very fast.

mwachs5, 2019-10-19:
`api=config-sifive` is a submodule. You need to initialize your submodules to get that directory. Please see the README: https://github.com/chipsalliance/rocket-chip#checkout-the-code

And make sure you do the last step`git submodule --init`

angi10, 2019-10-23:
@mwachs5 the submodule api-config-sifive doesn't exist in https://github.com/ucb-bar/rocket-chip.git

mwachs5, 2019-10-23:
Ah. I guess there is `api-config-chipsalliance`

Maybe try fixing this line: https://github.com/chipsalliance/rocket-chip/blob/65cf4921985093156196b47990a2d616bb40533a/Makefrag#L40

angi10, 2019-10-24:
I modified that line from `api-config-sifive/design/craft` in `api-config-chipsalliance/design/craft`
But the problem is: `find: "/home/andrea13/rocket-chip/api-config-chipsalliance/design/craft/src/main/scala": File or directory doesn't exist`
I have home/andrea13/rocket-chip/api-config-chipsalliance/design/craft/src/config/Config.scala

dsw, 2019-11-20:
Did this item get resolved?  I am having the same problem when running 
`rocket-chip/emulator$ make run-asm-tests-debug`

OmarAlsherbini, 2019-12-18:
The Makefrag file was updated yesterday to fix the api-config-sifive >> chipsalliance issue. Please refer back to the rocket-chip respiratory.

closed 2020-02-09 by mwachs5