chipsalliance/rocket-chip #2154: hart number specification for rocket core system

zohan180, 2019-10-17:
Hi, from reading the code and run the chip generator: I found that the harts number in Plic is 2, so 2 harts for this default config system, so external interrupts connect to both hart. However, when I check in CLINT module, the tile number is being for timing interrupts, and tile number is 1, so timing interrupts only to hart 0. I am confused here that why there is only one mtimecmp register for the system, is it should be 2 in this case for both hart on the RISC-V doc? Did I miss anything in this case? 

Also, I already used this config to run a firesim quad core config, in those simulation, seems like only 1hart/core, which agree with CLINT module but not PLIC. I tried to find these configs but have no luck with it. Where can i find these config and making sure it is doing what I think?

aswaterman, 2019-10-17:
There is only one hart in the default configuration. The PLIC has two
contexts per hart (one for M-mode and one for S-mode) whereas the CLINT
only has one context per hart (for M-mode).

HengZhuo2, 2019-10-17:
Thanks for the declaration. I was just confused by the way nHarts and harts were using in plic module assuming they are the same, did not realize they actually means contexts in this case.

closed 2019-10-17 by HengZhuo2

aswaterman, 2019-10-17:
Yeah, that is admittedly a bit confusing.