chipsalliance/rocket-chip #2158: Error while building C simulator

angi10, 2019-10-22:
I am following the instructions for building the Rocket Chip generator.
I have installed the Rocket-tools repository, Chisel3, and the RISCV environment.
 
Now, in the emulator directory, i type `make`
The first message is 

> find: "/home/andrea13/rocket-chip/api-config-sifive/design/craft/src/main/scala": File or directory doesn't exist
Then i have a lot of warnings and in the end it appears:

> Makefrag-verilator:42: instructions set for target "verilator/src/verilator-4.008/configure" not done
make: *** [verilator/src/verilator-4.008/configure] Error 2

Instead i have  api-config-chipsalliance, where can i find the other submodule?

CHNRailgun, 2019-10-28:
Same problem. Have you solved this problem?
Thank you!

angi10, 2019-10-28:
Just delete that directory at the line "default_submodules" in the Makefrag in rocket-chip directory.
Then i've just ignored that message and all the setup works.

CHNRailgun, 2019-10-28:
> Just delete that directory at the line "default_submodules" in the Makefrag in rocket-chip directory.
> Then i've just ignored that message and all the setup works.

thx a lot！

closed 2020-02-09 by mwachs5