chipsalliance/rocket-chip #2160: Simulation speed for rocket-chip: VCS vs Verilator

hyf6661669, 2019-10-23:
Can anyone tell me the the difference of simulation time between VCS and Verilator, for running rocket-chip? I have no license for VCS. It is said that verilator is the fastest opensource RTL simulator. And I want to know how fast can verilator be.The test maybe like this, https://github.com/chipsalliance/rocket-chip/pull/1351:
```
Yes, I do the benchmark on a i5-8600K with 4.1GHz Tubro, 16G DDR4, 3200MHz
which is a 6core machine, the benchmark command is

time make -j6 run-asm-tests
The result is, Verilator: 42 mins, iverilog: 2242mins
Which means Verilator is 50x faster then iverilog.
No much surprise, but I think standard verilog simulation is need by rocket-chip.
Since not everyone has VCS installed, iverilog is a common choice
(Another choice is oss-cvc , a much faster open source simulator, but I give up since it fails with internal error while compiling rocket-chip's verilog code)

I think the benefit of using Icarus Verilog is you can use it as a 4-state simulator and cross check the result with verilator's.
```

seldridge, 2019-10-23:
At a high level... and using the included Makefiles in `emulator` and `vsim`... Verilator takes longer to compile (3x as long for a serial build), but simulates faster (4--6x as fast). For really short test programs, VCS will do better due to its fast compile time. You can parallelize the Verilator build (`-jN`) to cut down on this.

It is not clear to me where this is coming from, however. My VCS is a couple years old, so newer versions may be better. Also, Verilator is, for Rocket Chip, default configured to build a 2-thread, multithreaded binary (you can change this with the [`VERILATOR_THREADS` option](https://github.com/chipsalliance/rocket-chip/blob/f072bef7aeb388987acafdbb9be97efcdf77db5e/emulator/Makefrag-verilator#L55) which changing this on my end was able to get a further simulation speedup). It's not clear to me how many threads VCS is utilizing for simulation.

The takeaway is that Verilator is likely good enough for most use cases unless there's a VCS feature that you really need. Or: there's not a 50x difference like with Icarus Verilog.

closed 2019-11-01 by hyf6661669

kev-cam, 2026-10-07:
A different simulator doing RTL -

https://github.com/kev-cam/sv2ghdl/blob/main/bfit/benchmarks/gpu_farm.md