chipsalliance/rocket-chip #2161: Use built-in async reset support

jackkoenig, 2019-10-24:
@hcook as requested in https://github.com/chipsalliance/rocket-chip/pull/2074#issuecomment-546010373

With https://github.com/chipsalliance/rocket-chip/pull/2074 we should use Chisel's built-in async reset support. This is a tracking issue for the steps to better utilize this feature. This is **not** about making rocket-chip use async reset instead of synchronous reset across the board, it's just cleaning up the code for the async reset flops we're already using.

TODO:
- [ ] Remove `src/main/scala/util/AsyncResetReg.scala` and uses of `AsyncResetReg` and `AsyncResetRegVec`
- [ ] Use `AsyncReset` type for async reset lines instead of `Bool`

This all I have off of the top of my head, will edit if there is more.

**Type of issue**: other enhancement

**Impact**: no functional change

**Development Phase**: proposal

mwachs5, 2019-10-24:
Plz add me as reviewer

mwachs5, 2019-10-24:
FYI @rmac-sifive ... this probably will interact with your PR https://github.com/chipsalliance/rocket-chip/pull/2159