chipsalliance/rocket-chip #2184: FSQRT_S and FSQRT_D's instruction decode: these 2 instructions needn't read rs2, but IDecode says "Y"

zhouqinghua, 2019-11-06:
**Type of issue**: bug report | feature request | other enhancement

**Impact**:  unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
as riscv-spec-v2.2's description, FSQRT_S and FSQRT_D floating instruction don’t need access rs2.
IDdecode should decode "rfs2" signal as "N", but the current code give "Y". it will increase some unnecessary RAW hazards in FPU's pipeline.

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2019-11-06:
I think those control signals are vestigial and aren't actually used.  The FPUDecoder, which provides this info to the hazard checker, is more precise about this.  So Rocket should not interlock on rs2 for fsqrt.

zhouqinghua, 2019-11-07:
thanks a lot. 
i take a look at rocket's code, rfs1/2/3 are not used. i think it is better to remove these unused signals.

closed 2019-11-07 by zhouqinghua