chipsalliance/rocket-chip #2185: ExampleRocketSystem mem_axi -- parameterized data width?

gsomlo, 2019-11-06:
I'm using Rocket as part of a LiteX (https://github.com/enjoy-digital/litex) SoC, which, depending on the physical FPGA dev board in use, has a DRAM controller whose axi-port data width may be either 64, 128, or 256 bits.

Couple of questions:
1. If I force Rocket's mem_axi data width to 128 or 256 by explicitly setting beatBytes, am I right to expect shorter and wider bursts when cache lines (which I _think_ are 512 bits) are fetched or flushed to main RAM? Would this qualify as an unquestionably "good thing"? (Intuitively, adding external-to-Rocket RTL to convert between mem_axi's default 64bit data width to something wider on the DRAM controller side should only add waste and complexity, but I'd like a sanity check on that intuition...
2. Would it make sense to parameterize the ExtMem port beatBytes, like in this patch:
```
diff --git a/src/main/scala/subsystem/Configs.scala b/src/main/scala/subsystem/C
onfigs.scala
index ceb29e48..cc56c6c4 100644
--- a/src/main/scala/subsystem/Configs.scala
+++ b/src/main/scala/subsystem/Configs.scala
@@ -348,6 +348,15 @@ class WithDefaultMemPort extends Config((site, here, up) =>
 {
                       idBits = 4), 1))
 })
 
+class WithLitexMemPort(dw: Int) extends Config((site, here, up) => {
+  case ExtMem => Some(MemoryPortParams(MasterPortParams(
+                      base = x"8000_0000",
+                      size = x"8000_0000",
+                      // FIXME: ensure 'dw' is a power of 2 (e.g., 1, 2, 4)
+                      beatBytes = site(MemoryBusKey).beatBytes * dw,
+                      idBits = 4), 1))
+})
+
 class WithNoMemPort extends Config((site, here, up) => {
   case ExtMem => None
 })
...
diff --git a/src/main/scala/system/Configs.scala b/src/main/scala/system/Configs.scala
index 2c81ddba..47e23826 100644
--- a/src/main/scala/system/Configs.scala
+++ b/src/main/scala/system/Configs.scala
@@ -74,3 +74,28 @@ class MMIOPortOnlyConfig extends Config(
 
 class BaseFPGAConfig extends Config(new BaseConfig)
 class DefaultFPGAConfig extends Config(new WithNSmallCores(1) ++ new BaseFPGAConfig)
+
+class BaseLitexConfig extends Config(
+  new WithLitexMMIOPort() ++
+  new WithNoSlavePort ++
+  new WithNExtTopInterrupts(4) ++
+  new BaseConfig
+)
+
+class LitexLinuxConfigW extends Config(
+  new WithNMedCores(1) ++
+ // "wide" (128bit) mem_axi
+  new WithLitexMemPort(2) ++
+  new BaseLitexConfig
+)
+class LitexLinuxConfigDW extends Config(
+  new WithNMedCores(1) ++
+ // "double-wide" (256bit) mem_axi
+  new WithLitexMemPort(4) ++
+  new BaseLitexConfig
+)
...
```
3. Finally, and this is probably more of a Chisel question, is there a simple way to ensure that the `dw` Int argument to `WithLitexMemPort` is either 1, 2, 4, or max. 8, but not anything other than a power-of-two in that range (if there's an example of that sort of Chisel code anywhere within the Rocket sources, I'd appreciate a pointer! :)

hcook, 2019-11-08:
1. Yes, you should expect shorter and wider bursts of the same total size. 64B is the default global setting, which is controlled via the `CacheBlockBytes` `Field`.
2. Whether the port width setting should drive the width of the bus, or whether the bus width setting should drive the width of the ports attached to it, or whether the two are free variables is a matter of policy, which is why we leave it up to the `Config`uration . The default policy is that configuring the memory bus to be wider will make all memory ports wider. You can make the memory bus wider by changing the `beatBytes` argument of the `MemoryBusKey` to the width that you want:
```
class WithLitexMemPort(dw: Int) extends Config((site, here, up) => {
   case MemoryBusKey => up(MemoryBusKey, site).copy(beatBytes = dw)
}
```
Of course, you could instead change the policy to make the parameter values totally independent from one another:
```
class WithLitexMemPort(dw: Int) extends Config((site, here, up) => {
   case ExtMem => Some(MemoryPortParams(MasterPortParams(..., beatBytes = dw)))
}
```
... but you might end up with a wide port attached to a narrow bus, which is probably not what you actually want. (Multiplying one byte width by another byte width as you did in your specific example does not seem correct.)
3. `chisel3.util.isPow2(dw) && dw <=8`

gsomlo, 2019-11-10:
>You can make the memory bus wider by changing the `beatBytes` argument of the `MemoryBusKey` to the width that you want:
> ```
> class WithLitexMemPort(dw: Int) extends Config((site, here, up) => {
>    case MemoryBusKey => up(MemoryBusKey, site).copy(beatBytes = dw)
> }
> ```

If I do that, my `mmio_axi` port will also be wider, and that's not what I need either, as the MMIO bus connecting me to my peripherals' control registers is not (necessarily) the same width as the datapath of my DRAM controller.

> Of course, you could instead change the policy to make the parameter values totally independent from one another:
>
> ```
> class WithLitexMemPort(dw: Int) extends Config((site, here, up) => {
>    case [...]

Can you please elaborate on this in a bit more detail? I've been doing this exact thing with my original patch, getting wider (2x or 4x `mem_axi` ports), and "standard" width (64-bit) `mmio_axi` ports, but since I'm not touching `MemoryBusKey`, am I indeed attaching a wide `mem_axi` port to a narrow bus? What am I missing (and potentially messing up) here? :)

>(Multiplying one byte width by another byte width as you did in your specific example does not seem correct.)

`dw` was a scalar multiplier (1, 2, or 4) applied to the original `site(MemoryBusKey).beatBytes`, so apart from the parameter being perhaps improperly _named_, the functionality was correct (modulo my immediately preceding question, that is).

Thanks!

hcook, 2019-11-11:
> If I do that, my mmio_axi port will also be wider, and that's not what I need either, as the MMIO bus connecting me to my peripherals' control registers is not (necessarily) the same width as the datapath of my DRAM controller.

Hmm, yes, the coupling of those parameters in the example configuration is weird and I would probably call it a bug. The ports are attached to different crossbars, whose widths can be controlled independently. Widening one port does not have to widen the other. 

> am I indeed attaching a wide mem_axi port to a narrow bus?

Yes. I recommend widening the width of the SBus and MBus to be as wide as you want the memory port to be.

> would still issue a burst of minimum 2 beats (for a total of 64 bits) even when the CPU only issues a 32bit memory access (and therefore one of the two beats would presumably be strobed out completely)?

The behavior of memory accesses and MMIO accesses is different. The DCache will fetch entire cache blocks from memory regardless of the original load or store size. For an MMIO accesses, Rocket will only request as much data as it needs to satisfy the request.

gsomlo, 2019-11-11:
Thanks! With the understanding that `WithLitexMMIOPort` should _not_ reference `MemoryBusKey`, it's now obvious that widening the memory bus to the desired data width of `mem_axi` is the "canonical" solution I was looking for :)

That said, there are a few identically set up "BusKeys" in subsystem/Configs.scala, in addition to `Memory`: `System`, `Control`, `Periphery`, and `Front`. For an MMIO port, I'd have expected `Periphery` to apply, but then I'm not familiar enough with Rocket internals to give this feeling any strength at all :)

Finally,

>For an MMIO accesses, Rocket will only request as much data as it needs to satisfy the request.

my experiments show that this seems true even for the case where I narrow `mem_axi` down to less than 64 bits (XLen). At 32-bit `mmio_axi` width 32-bit (or narrower) loads or stores involving an MMIO region will manifest as single-beat 32-bit operations, rather than a pair of 32-bit beats totaling 64 bits, as I expected (and feared), given that the CPU is a 64bit one.

Finally, I tried enforcing the data width via a `require` statement like so:
```
class WithLitexMemPort(dataBits: Int) extends Config((site, here, up) => {
  require(dataBits == 64 || dataBits == 128 || dataBits == 256)
  case MemoryBusKey => up(MemoryBusKey, site).copy(beatBytes = dataBits/8)
  case ExtMem => Some(MemoryPortParams(MasterPortParams(
                      base = x"8000_0000",
                      size = x"8000_0000",
                      beatBytes = site(MemoryBusKey).beatBytes,
                      idBits = 4), 1))
})
```
but that's a syntax error, on the `MemoryBusKey` line (`}` expected but `case` found). My Chisel is not strong enough to figure out if this is just not the right place for a `require`, or if there's some additional syntactic sugar I'm missing...

Either way, thanks for all your help, and please feel free to close the issue (I've got it bookmarked for future reference)!

hcook, 2019-11-11:
Oh I forgot about `WithEdgeDataBits`... it is definitely relevant to your use case, though I'm not sure whether the original designer intention was that it change the MMIO width or not. I might need to do a little more detective work w.r.t. that intention before merging #2187...

The hierarchical crossbar topology is not especially well documented but you can see it being created rather straightforwardly in
https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/subsystem/BusTopology.scala
Mediocre ASCII art rendition:
```
FrontBus  =>  |                    | => ControlBus => PeripheralBus
                        |SystemBus | => (MMIO Port)
Cores        => |                    | => Memory Bus => (MemoryPort)
```

If you take a look in https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/subsystem/Ports.scala I think you can imagine adding traits that add new ports that are coupled to different buses instead. The current ones are kind of intended as examples moreso than as an exhaustive set of options.

>  I tried enforcing the data width via a require statement

This is just a scala syntax issue. You could either wedge it in between `Config((site,` like `Config({ require(...); (site...) => { ... } })` or in one of the cases like `case ExtMem => { require(...); Some(...) }`. The latter will probably look nicer but the former is safer because the latter will result in only checking the requirement when one particular field is looked up (unless you add it to all fields).

> my experiments show that this seems true even for the case where I narrow mem_axi down to less than 64 bits (XLen). At 32-bit mmio_axi width 32-bit (or narrower) loads or stores involving an MMIO [...]

I just want to make sure I understand your scenario: A 64b core is issuing a 32b `lw`/`sw` to a 32b port, and you are expecting it to appear on the interface as a single 2x32b transaction? I would only expect a single beat if the data only fits in a single beat. The only way to provoke a 2 beat transaction would be to use `ld`/`sd` to ask for 64b at once. (That 64b access should show up as a 2beat*32b transaction, not as 2 separate 32b transactions, let me know if that is not what you observe because that would be unexpected.)

gsomlo, 2019-11-12:
Thanks, this was all really helpful!

> I just want to make sure I understand your scenario: A 64b core is issuing a 32b `lw`/`sw` to a 32b port, and you are expecting it to appear on the interface as a single 2x32b transaction? I would only expect a single beat if the data only fits in a single beat.

I'd expect a 64bit core to always issue 64bit loads and stores; the lb/sb/lw/sw ones would simply have a lot of byte strobes set to 'disable'. Now, there's either a (L1) cache inbetween that and actual RAM, and I get that interaction between those occurs in whole cache lines at a time. I was wondering if the AXI logic sitting inbetween core initiated loads/stores and the MMIO port would optimize away the strobed-out bytes if the port width is narrower than XLen, and I'm happy to see it does :)

>The only way to provoke a 2 beat transaction would be to use `ld`/`sd` to ask for 64b at once. (That 64b access should show up as a 2beat*32b transaction, not as 2 separate 32b transactions, let me know if that is not what you observe because that would be unexpected.)

What I'm observing is consistent with what you describe as the correct, expected behavior. Sorry for the confusion :)

closed 2019-11-13 by hcook