chipsalliance/rocket-chip #2190: Synthesizing to FPGA

gdessouky, 2019-11-12:
When I generate Verilog to target FPGA, a 'ram' reg is generated (plz see below) which is much too large to fit in any typical FPGA BRAMS during synthesis. Is that normal or a glitch of some sort at my end? I reduce it extremely just to see synthesis through, but I am not really sure what am I doing and what does it affect. 

module mem_ext(
  input W0_clk,
  input [24:0] W0_addr,
  input W0_en,
  input [63:0] W0_data,
  input [7:0] W0_mask,
  input R0_clk,
  input [24:0] R0_addr,
  input R0_en,
  output [63:0] R0_data
);

  reg reg_R0_ren;
  reg [24:0] reg_R0_addr;
  reg [63:0] ram [33,554,431:0];    //this is much too large!
  `ifdef RANDOMIZE_MEM_INIT
    integer initvar;
    initial begin
      #`RANDOMIZE_DELAY begin end
      for (initvar = 0; initvar < 33554432; initvar = initvar+1)
        ram[initvar] = {2 {$random}};
      reg_R0_addr = {1 {$random}};
    end
  `endif

gdessouky, 2019-11-12:
Also, SimDTM.v which is instantiated also generates syntax errors and commenting out the instantiation probably causes the entire design to get optimized out and I get zero utilization when synthesis completes. What is SimDTM.v (it imports a Cpp file)? And how is it instantiated in synthesizable Verilog?

hcook, 2019-11-12:
Sounds like you are targeting the testharness, its large test memory, and its simulation-only debug transport driver. Make sure you are selecting the actual design top level module and not the testharness module for mapping to FPGA.

sanket2608, 2020-01-15:
I'm curious about how you are going to verified your design,means where are you going to put your processor code. If you are commenting out SimDTM and making testharness as top module so synthesis tools interpret as nothing is going in and coming out as you can see in testharness only two input clock ,reset and a output io_success which is connected to sim_exit.
So it is optimizing whole circuit.

zhh124, 2020-04-09:
If you want to directly map TestHarness into FPGA, you need to modify SimDTM module, and reduce external sram size, gsomlo had did these in his/her project:
https://github.com/gsomlo/yoloRISC