chipsalliance/rocket-chip #2200: Default Slave Port name changed (now prefixed with "ValName")

gsomlo, 2019-11-20:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request | proposal

**Other information**
 
The slave AXI port has been renamed from `l2_frontend_bus_axi4_0_*` to ` ValNamel2_frontend_bus_axi4_0_*`, which breaks any attempt to use `ExampleRocketSystem` from other projects. This started happening since somewhere approximately commit 5c78478e0568b9971352635f426858a648e352c9 (9 or 10 commits behind master, at the time of this writing).

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
Building RocketChip with the `WithDefaultSlavePort()` config option should reproduce the problem (see `ExampleRocketSystem` module in the generated verilog file)

**What is the expected behavior?**
slave AXI port name should stay unchanged (i.e., not be prefixed by the string `ValName`).

**What is the use case for changing the behavior?**
ability to link `ExampleRocketSystem` module from other projects

hcook, 2019-11-20:
Probably an unintended result of https://github.com/chipsalliance/rocket-chip/pull/2178

aswaterman, 2019-11-20:
Ah, I guessed it might've been caused by updating chisel, but that sounds likely.

commit 11b60ebd (2019-11-20): diplomacy: SourceNode.makeIOs uses ValName.name

Turns out `ValName.toString` prefixed the name with the string "ValName". `ValName.name` is what I wanted and even what I used in the SinkNode already.

Closes #2200
PR #2201 (2019-11-20): diplomacy: SourceNode.makeIOs uses ValName.name
Turns out `ValName.toString` prefixed the name with the string "ValName". `ValName.name` is what I wanted and even what I used in the SinkNode already.

**Related issue**: Closes #2200

**Type of change**: bug fix

**Impact**: bug fix

**Development Phase**: implementation
commit b5ed10e8 (2019-11-20): diplomacy: SourceNode.makeIOs uses ValName.name (#2201)

Turns out `ValName.toString` prefixed the name with the string "ValName". `ValName.name` is what I wanted and even what I used in the SinkNode already.

Closes #2200
closed 2019-11-20 by hcook