chipsalliance/rocket-chip #2202: How to evaluate the performance of rocket-chip?

hyf6661669, 2019-11-21:
I'm using verilator to run dhrystone(https://github.com/riscv/riscv-tests/tree/master/benchmarks/dhrystone) and get the following result:
```
Microseconds for one run through Dhrystone: 481
Dhrystones per Second:                      2074
mcycle = 241070
minstret = 199268

```
My question is:
1. How to convert the number to DMIPS/MHz? Since I don't know the frequency of verilator simulation.
2. How to compare the performance with ARM cpus, such as cortex A5/A7/M4/M7?
3. It is said dhrystone benchmark is out-dated and the result is influenced by the C compiler. Coremark is said ti be more effective for performance evaluation. Could you tell me how to run coremark in rocket-chip?

aswaterman, 2019-11-21:
The magic constant of 569 cycles per iteration corresponds to 1 DMIPS/MHz,
so 481 corresponds to 1.18 DMIPS/MHz.

Rocket is known to be able to get a Dhrystone score over 1.7 DMIPS/MHz, but
this depends heavily on the compiler and C library setup.  For example,
without an optimized strcmp implementation, performance will suffer quite a
bit.

hyf6661669, 2019-11-21:
> The magic constant of 569 cycles per iteration corresponds to 1 DMIPS/MHz, so 481 corresponds to 1.18 DMIPS/MHz. Rocket is known to be able to get a Dhrystone score over 1.7 DMIPS/MHz, but this [...]

If I use the same C compiler/library implementation/compile options, to compile the dhrystone benchmark and run the binary file on different risc-v cores, is it resonable to compare the performance of these cores by comparing their dhrystone results?

aswaterman, 2019-11-21:
That's a reasonable thing to do if you believe that Dhrystone is a good
proxy for the software you really want to run.  Occasionally, it is, but
more often it is not.  If you care about a particular application domain,
you should consider writing a microbenchmark that accurately reflects your
workload.

closed 2019-11-21 by hyf6661669