chipsalliance/rocket-chip #2211: Loading Rocket Core on Zedboard by Vivado

angi10, 2019-11-29:
Hello everyone,
my goal is to build Rocket Core on the Zedboard. 
I have generate the Rocket Core with DefaultSmallConfig, and i have the output files:

freechips.rocketchip.system.DefaultSmallConfig.0x0.0.regmap.json
freechips.rocketchip.system.DefaultSmallConfig.0x0.1.regmap.json
freechips.rocketchip.system.DefaultSmallConfig.0x2000000.0.regmap.json
freechips.rocketchip.system.DefaultSmallConfig.0x40.0.regmap.json
freechips.rocketchip.system.DefaultSmallConfig.0xc000000.0.regmap.json
freechips.rocketchip.system.DefaultSmallConfig.anno.json
freechips.rocketchip.system.DefaultSmallConfig.behav_srams.v
freechips.rocketchip.system.DefaultSmallConfig.conf
freechips.rocketchip.system.DefaultSmallConfig.d
freechips.rocketchip.system.DefaultSmallConfig.dts
freechips.rocketchip.system.DefaultSmallConfig.fir
freechips.rocketchip.system.DefaultSmallConfig.graphml
freechips.rocketchip.system.DefaultSmallConfig.json
freechips.rocketchip.system.DefaultSmallConfig.memmap.json
freechips.rocketchip.system.DefaultSmallConfig.plusArgs
freechips.rocketchip.system.DefaultSmallConfig.rom.conf
freechips.rocketchip.system.DefaultSmallConfig.v

I tried creating a new project in Vivado, adding as sources the two verilog files ...DefaultSmallConfig.v and ...DefaultSmallConfig.behav_srams.v plus the other files required like pulsarg_reader.v, AsyncResetReg.v and EICG_wrapper.v

First of all, i have a syntax error in DefaultSmallConfig.behav_srams.v in line 221, the message is:
integer overflow when computing array size

hyf6661669, 2019-12-03:
Hi, what is the content of DefaultSmallConfig.behav_srams.v

aswaterman, 2019-12-03:
The error you're seeing comes from the simulated main memory in the test harness; it is not meant to be synthesized.  Can you try setting the top-level module in your synthesis experiment to ExampleRocketSystem and see if the problem goes away?

In the longer term, we should consider splitting the Verilog for the DUT from that of the test harness so that the synthesis tool never sees these other modules.

angi10, 2019-12-03:
You're right, i've set the module ExampleRocketSystem as top module and the elaboration has been done.

I've another question, is DefaultSmallConfig.v file the verilog description only of the rocket core or of the entire system? 
I post below some images to help you understanding what i'm looking to know.

[image]
Is that file a description of the Rocket core displayed in tile2, or is a description of the system including the networking to external memories?
Thanks

aswaterman, 2019-12-03:
Cool.

It’s a description of a “core complex”, i.e., the Rocket core, caches/TIMs,
MMIO network, interrupt controller, and AXI bridge.

angi10, 2019-12-04:
Thanks a lot!

closed 2020-02-09 by mwachs5

Uzair540, 2020-10-28:
> You're right, i've set the module ExampleRocketSystem as top module and the elaboration has been done.
>
> I've another question, is DefaultSmallConfig.v file the verilog description only of the rocket [...]
@angi10 
How you set the module ExampleRocketSystem as top module.I am having same issue whil synthesizing
[image]

jerryhethatday, 2020-12-31:
> The error you're seeing comes from the simulated main memory in the test harness; it is not meant to be synthesized. Can you try setting the top-level module in your synthesis experiment to [...]

> In the longer term, we should consider splitting the Verilog for the DUT from that of the test harness so that the synthesis tool never sees these other modules.
Have you already done this?( splitting the Verilog for the DUT from that of the test harness??)