chipsalliance/rocket-chip #2218: 64bits DTIM configuration run fail

roychen1998, 2019-12-06:
Hi all,

I am configuring the large core to use DTIM instead of DCache by adding "scratch=Some(0x20000000)" to DCache parameters.
When I run asm tests, it passed many cases then failed at case "rv64ua-p-amoadd_w.riscv";
And when I run bmark tests, it failed when running case "mm.riscv".
To be compared, the Tiny core with DTIM run rv32ua-p-amoadd_w and mm.riscv pass.

Did someone ever try DTIM on 64bits configuration?

Best Regard,
Roy

aswaterman, 2019-12-06:
Might need a larger DTIM?

roychen1998, 2019-12-08:
Thanks aswaterman for your suggestion. Here is some more information for my large core:

Multi-threads cases all fail. 
64 bits atomic tests rv64ua-p all fail, but rv32ua-p cases pass
64 bits rv64si-p fail, rv32si-p pass
All "-v-" tests fail.
Other cases can pass.

Have got some idea?

Best Regard,
Roy

roychen1998, 2019-12-09:
I found that in TLB.scala for the following statements:

val lrscAllowed = Mux(Bool(usingDataScratchpad ...
    ...
Mux(Bool(usingAtimics) && io.req.bits.cmd.isOneOf(M_XLR, M_XSC), ~lrscAllowed, 0.U)

Does it mean atomic will not be supported when DTIM is in-used?
I am configuring a new toolchain without atomic and will try the test again.

Best Regard,
Roy

roychen1998, 2019-12-09:
I found that mm.c in benchmark is depended on atomic instruction of __sync_fetch_and_add_4. So the root cause is for atomic. 
But according to the code in rocket, it seems that only LR and SC are not supported. I am adding some print and dumping the waveform to figure out why rv64ua-p-amoadd_w fail.

roychen1998, 2019-12-09:
In the waveform a instructure trying to write satp is regarded as illegal instruction.
0x180 SRW satp Supervisor address translation and protection
what is this satp register for?

aswaterman, 2019-12-09:
“Address translation and protection.” It controls  The virtual memory
configuration.

roychen1998, 2019-12-12:
I have looked into the waveform and found something:
For a normal rv64ua-p-amoadd_w operation, replay_wb will be signaled and the pc will be pushed and that pc will be poped several cycles later.
I found that replay_wb is asserted for io_dmem_s2_nack get high. 

But for DTIM case, the pc just jump to csr_io_evec and  rv64ua-p-amoadd_w will never be execuated.

Is this the supposed behavior for DTIM?

Best Regard,
Roy

aswaterman, 2019-12-12:
The exception could be caused by any number of things. I am guessing it’s an access exception caused by an invalid address because your DTIM isn’t big enough.

roychen1998, 2019-12-13:
Hi aswaterman,

The root cause is found that I was trying to using atomic instuction to a non-cacheable region.

Best Regard,
Roy

aswaterman, 2019-12-13:
Oh, I see, that will do it!

closed 2019-12-13 by aswaterman