chipsalliance/rocket-chip #2231: Rocket Stage

seldridge, 2019-12-19:
The new `Stage`/`Phase` framework is available with an updated `DependencyAPI` coming in FIRRTL 1.3. This issue covers migrating the existing `freechips.rocketchip.system.Generator` object to a new `RocketStage` consisting of phases derived from the existing steps used to build `RocketChip`. 

See: https://github.com/chipsalliance/rocket-chip/blob/1cf70ea21810887828be70da8239366c97feb0cc/src/main/scala/system/Generator.scala#L106

This would benefit from support for a `RunPhaseAnnotation` that would then allow injection of new phases into the Rocket Chip generation process.

This is intended to be command-line compatible with Rocket Chip's existing positional arguments (https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/util/GeneratorUtils.scala#L80). A follow-on PR could move these to something with actual arguments as opposed to positions.

Random cool thing with this is that it would allow you to build a Rocket Chip fat jar that could support simple core configurations, e.g., :

```bash
./utils/bin/rocket-chip --cores 2 --core-uarch boom --roccs hwacha,gemmini
```

Or that could be used as a command line interface to build an anonymous context dependent environment config stack:

```bash
./utils/bin/rocket-chip --config WithRV32,DefaultConfig
```

**Type of issue**: feature request

**Impact**: API modification

**Development Phase**: proposal

davidmlw, 2023-12-05:
It is a cool feature. Will rocketchip get back to this idea?

sequencer, 2023-12-05:
No, rocketchip will be split to smaller libraries, and user should handle the build flow on their own.