chipsalliance/rocket-chip #2242: Some question for mechanism of update BTB at rocket-chip

linkLiao, 2020-01-07:
There are two way to Update BTB, one from Pipeline, another from Frontend.
I find the code to update BTB at Fronend Module like:

``` scala
    when (!io.cpu.btb_update.valid) {
      val fetch_bubble_likely = !fq.io.mask(1)
      btb.io.btb_update.valid := fq.io.enq.fire() && !wrong_path && fetch_bubble_likely && updateBTB
      btb.io.btb_update.bits.prediction.entry := UInt(tileParams.btb.get.nEntries)
      btb.io.btb_update.bits.isValid := true
      btb.io.btb_update.bits.cfiType := btb.io.ras_update.bits.cfiType
      btb.io.btb_update.bits.br_pc := s2_base_pc | (taken_idx << log2Ceil(coreInstBytes))
      btb.io.btb_update.bits.pc := s2_base_pc
    }
```
But Signal *fq.io.enq.valid* may be high when signal *icache.io.resp.valid* is low

So, what if the garbage instruction in this cycle exist jump or branch instruction, and orignal instruction don't exist these control instruction.

The BTB will be updated and Pipeline will not correct it.

But make me confuse is that, rocket-chip work as usually, 
Is the BTB has other mechanism to correct these wrong operation.

aswaterman, 2020-01-07:
The Rocket pipeline itself validates the PC of each instruction.  So if the BTB is populated with garbage, it has only a performance impact, not a correctness impact.

linkLiao, 2020-01-10:
But when BTB update garbage info, ICache will send some garbage request into memory, 
In SoC design, I may just use part of full address space,
if  the request into the memory space that I don't use, and I don't implement the response mechanism for these memory space don't use, 
it will be into deadlock because of ICache send a memory request but no one send response.

aswaterman, 2020-01-10:
Not responding to messages is illegal behavior in TileLink.

closed 2020-02-09 by mwachs5