chipsalliance/rocket-chip #2302: Can't add a module to the system

angi10, 2020-02-21:
Hi everyone,
my goal is to add a new module to the examplerocketsystem.
I generate a configuration with DefaultSmallConfiguration + WithJtagDTM and i have the verilog file.
Now i want to add another module into the system, but i can't do it.
When i modify the src files of the rocket chip generator, it remains equal to the previous configuration.
Please, help me

mwachs5, 2020-02-21:
how did you add the module to the system?

angi10, 2020-02-21:
Thanks for the fast answer.
I simply added these lines
`class example extends Module {
val io = IO(new Bundle {
  val example_in=Input(Bool())
  val example_out=Output(Bool())
})
io.example_out := io.example_in`

when i run, it generates the .v file in generated-src but there isn't any trace of that module

angi10, 2020-02-21:
I added these code in a generic file which is loaded by the system

mwachs5, 2020-02-21:
you need to actually instantiate your module somewhere:

```
val my_example = Module(new example())

printf(s"Example output is : ${my_example.io.example_out}")

```

angi10, 2020-02-21:
Ok, i'm trying it. I'll let you know if i resolve.
Thanks a lot

angi10, 2020-02-21:
`[error] Caused by: chisel3.internal.ChiselException: Error: Called Module() twice without instantiating a Module`

angi10, 2020-02-21:
I've created the class example and i have insert it in the `package freechips.rocketchip.subsystem`.
Then i have instantiate that module in `class RocketSubsystem`