chipsalliance/rocket-chip #2304: HasAPBControlRegMap hangs processor

sevens-elevens, 2020-02-23:
**Type of issue**: bug report

**Impact**:  API modification

**Development Phase**: request 

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
- Create a class that extends RegisterRouter with HasAPBControlRegMap. 
- In that class, create an AXI4 node and APB node (not sure if this is needed). 
- Use the **regmap** function to assign a memory addressable register.
- Connect all three nodes to the memory bus with **mbus.toDRAMController**.
- The APB register is connected to the controlNode defined in the trait, like so:
```scala
ddr4.controlNode :=  mbus.toDRAMController(Some("ddr4_mem_ctl_apb")) {
        TLToAPB() :=
          TLDelayer(0.2) :=
          TLBuffer(BufferParams.flow) :=
          TLDelayer(0.2) :=
          TLFragmenter(minSize = 8, maxSize = 64, holdFirstDeny = true)
```
 - The processor hanging happens with just TLToAPB in the connection path, with just TLToAPB and the TLFragmenter in the connection path, and a few other combinations. 

**What is the current behavior?**
- The code elaborates successfully but the processor hangs during simulation. 
- With all other things being held constant, the simulation executes successfully when in the class definition, **HasAPBControlRegMap** is replaced with **HasAXI4ControlRegMap** and the register node connected to the controlXing defined in the trait, like so: 
```scala
    ddr4.controlXing(SynchronousCrossing()) :=
      mbus.toDRAMController(Some("ddr4_mem_ctl_axi4")) {
        AXI4UserYanker() :=
          AXI4IdIndexer(0) :=
          TLToAXI4() :=
          TLFragmenter(minSize = 8, maxSize = 64, holdFirstDeny = true)
      }
```
- When the processor hangs, it gets stuck on an infinite loop of this:
```bash
r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35017 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35018 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35019 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35020 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35021 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35022 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35023 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35024 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35025 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
C0:      35026 [x] pc=[00000045a2c41045] W[r 0=0000000000000000] R[r12=0000000000000000] R[r30=0000000000000000] inst=[00004231] DASM(00004231)
```

**What is the expected behavior?**
The processor should not hang. 

**Please tell us about your environment:**

- Version: 4f0cdea8
- OS: Linux 2.6.32-754.11.1.el6.x86_64 #1 SMP Tue Jan 22 17:25:23 EST 2019 x86_64 x86_64 x86_64 GNU/Linux

**What is the use case for changing the behavior?**
To make it so a processor doesn't hang when using this utility. Additionally, helping me understand why this happens.

mwachs5, 2020-02-23:
Can you produce a waveform of traffic hitting the TL2APB? Are you expecting traffic to go to the TL2APB?

sevens-elevens, 2020-02-24:
Thanks for the quick reply. The test we're running doesn't try to talk to anything on the APB bus. It's talking to an SRAM on another AXI4 bus. It looks like there's still activity, but the simulation stalls and nothing is moving through the instruction pipeline. 

[image]
[image]
[image]