chipsalliance/rocket-chip #2321: Build (verilog) fails if "CONFIG=..." specified on makefile commandline

gsomlo, 2020-03-05:
**Type of issue**: bug report

**Impact**:  unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
```
git clone https://github.com/chipsalliance/rocket-chip
cd rocket-chip
make RISCV=${HOME}/RISCV -C vsim verilog CONFIG=DefaultConfig
```

**What is the current behavior?**
```
[info] running freechips.rocketchip.system.Generator -td /home/somlo/rocket-chip/vsim/generated-src -T freechips.rocketchip.system.TestHarness -C DefaultConfig
[error] (run-main-0) java.lang.Exception: Unable to find part "DefaultConfig" from "ArrayBuffer(DefaultConfig)", did you misspell it?
[error] java.lang.Exception: Unable to find part "DefaultConfig" from "ArrayBuffer(DefaultConfig)", did you misspell it? 2s
[error] 	at Chisel.package$throwException$.apply(compatibility.scala:423)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.$anonfun$getConfig$1(GeneratorUtils.scala:19)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight(IndexedSeqOptimized.scala:65)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight$(IndexedSeqOptimized.scala:72)
[error] 	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:49)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig(GeneratorUtils.scala:14)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig$(GeneratorUtils.scala:13)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.getConfig(PreElaboration.scala:16)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:26)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:16)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:32)
[error] 	at firrtl.options.phases.DeletedWrapper.internalTransform(DeletedWrapper.scala:15)
[error] 	at firrtl.options.Translator.transform(Phase.scala:140)
[error] 	at firrtl.options.Translator.transform$(Phase.scala:140)
[error] Caused by: java.lang.ClassNotFoundException: DefaultConfig
[error] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:382)
[error] 	at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:92)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:419)
[error] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:352)
[error] 	at java.lang.Class.forName0(Native Method)
[error] 	at java.lang.Class.forName(Class.java:264)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.$anonfun$getConfig$1(GeneratorUtils.scala:16)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight(IndexedSeqOptimized.scala:65)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight$(IndexedSeqOptimized.scala:72)
[error] 	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:49)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig(GeneratorUtils.scala:14)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig$(GeneratorUtils.scala:13)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.getConfig(PreElaboration.scala:16)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:26)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 48 s, completed Mar 5, 2020 11:49:58 AM
```
**_NOTE_**: if `CONFIG=DefaultConfig` is left out, the verilog for the default config is built without problems. However, I would like to build something **_other_** than the `DefaultConfig`, and can't select it on the `make` command line as things stand right now!

**What is the expected behavior?**

Verilog for the selected configuration is built without error.

**Please tell us about your environment:**

Fedora 30, x86_64

**What is the use case for changing the behavior?**
Selecting a configuration **_other_** than `DefaultConfig`.

seldridge, 2020-03-05:
I believe this needs the full package name, e.g., `CONFIG=freechips.rocketchip.system.DefaultConfig`.

chengleilucas, 2020-03-05:
Could you try add SUB_PROJECT=rocketchip to the make command?

gsomlo, 2020-03-05:
Adding `SUB_PROJECT=rocketchip` results in the same error (and so does adding `CFG_PROJECT=rocketchip`).

Spelling it out as `CONFIG=freechips.rocketchip.system.DefaultConfig` seems to work though -- is this the new "canonical" way of doing things ?

FTR, bisect says the change was introduced by commit `9d997506bffa1971aa692cd304792b969dfeaf47`, "separate generator behavior into phases (#2274)".

seldridge, 2020-03-05:
Yeah, that changed with #2274. Specifically see the top level [`Makefrag`](https://github.com/chipsalliance/rocket-chip/pull/2274/files#diff-81cdb367f66ba3495d13610cde9eb7e1R9). @debs-sifive can probably comment more there.

The README and other documentation likely needs to be updated. Sorry about that!

debs-sifive, 2020-03-05:
Ah yes, my bad! You do need to specify the entire project/config for `CONFIG` now — I will update the READMEs ASAP.

gsomlo, 2020-03-05:
This new information puts me back in business. Feel free to close this issue if/when the README has been updated.

Thanks again to all of you!

debs-sifive, 2020-03-06:
Updated — thanks for raising this, @gsomlo!

closed 2020-03-06 by debs-sifive

gsomlo, 2020-03-29:
looks like as of right now (commit 38c2903) things are back to the old behavior (`CONFIG=DefaultConfig`, i.e. without the prepended `freechips.rocketchip.system`). Is the revert temporary, i.e. is the "new" behavior likely to be put back in place? I have a downstream project depending on Rocket, and would like to know whether to wait it out or whether to add my own revert of the `freechips.rocketchip.system` prepend?

Thanks much, as usual!

mwachs5, 2020-03-30:
Yes the intention is to keep the “new” behavior within a week or two