chipsalliance/rocket-chip #2323: Behavior mismatch between Rocket chip and Spike

JaewonHur, 2020-03-06:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Recently I've been verifying Rocket Chip using ISA simulator (Spike), comparing 
the machine status (GPRs and memory) after certain stream of instructions. 
It seems there is a mismatch.
I don't know whether it is bug or not, but could anyone check this behavior?

The detailed behavior is as follow.
First Rocket Chip atomically write a word (or double word) to some address using amo instructions (such as amoadd, amoswap ..). 
Then execute fence.i instruction and then jump to that address.
When doing this, Rocket chip just execute staled instruction (the value before atomic write).

You can reproduce it using below instruction stream.
I'm simulating with default config rocket chip.

```
  00000093        | addi       r1 ,r0 ,0x000
  00000113        | addi       r2 ,r0 ,0x000
  00000193        | addi       r3 ,r0 ,0x000
  00000213        | addi       r4 ,r0 ,0x000
  00000293        | addi       r5 ,r0 ,0x000
  00000313        | addi       r6 ,r0 ,0x000
  00000393        | addi       r7 ,r0 ,0x000
  00000413        | addi       r8 ,r0 ,0x000
  00000493        | addi       r9 ,r0 ,0x000
  00000513        | addi       r10,r0 ,0x000
  00000593        | addi       r11,r0 ,0x000
  00000613        | addi       r12,r0 ,0x000
  00000693        | addi       r13,r0 ,0x000
  00000713        | addi       r14,r0 ,0x000
  00000793        | addi       r15,r0 ,0x000
  00000813        | addi       r16,r0 ,0x000
  00000893        | addi       r17,r0 ,0x000
  00000913        | addi       r18,r0 ,0x000
  00000993        | addi       r19,r0 ,0x000
  00000a13        | addi       r20,r0 ,0x000
  00000a93        | addi       r21,r0 ,0x000
  00000b13        | addi       r22,r0 ,0x000
  00000b93        | addi       r23,r0 ,0x000
  00000c13        | addi       r24,r0 ,0x000
  00000c93        | addi       r25,r0 ,0x000
  00000d13        | addi       r26,r0 ,0x000
  00000d93        | addi       r27,r0 ,0x000
  00000e13        | addi       r28,r0 ,0x000
  00000e93        | addi       r29,r0 ,0x000
  00000f13        | addi       r30,r0 ,0x000
  00000f93        | addi       r31,r0 ,0x000
  300020f3        | csrrs      r1 ,mstatus,r0
  0080e093        | ori        r1 ,r1 ,0x008
  30009073        | csrrw      r1 ,mstatus,r0 
  304020f3        | csrrs      r1 ,mie,r0 
  00106113        | ori        r2 ,r0 ,0x001
  00b11113        | slli       r2 ,r2 ,0x00b
  0020e0b3        | or         r1 ,r1 ,r2
  30409073        | csrrw      r1 ,mie,r0
  305020f3        | csrrs      r1 ,mtvec,r0
  00606113        | ori        r2 ,r0 ,0x006
  01c11113        | slli       r2 ,r2 ,0x01c
  00116113        | ori        r2 ,r2 ,0x001
  0020e0b3        | or         r1 ,r1 ,r2
  30509073        | csrrw      r1 ,mtvec,r0
  00800193        | addi       r3 ,r0 ,0x008
  01c19193        | slli       r3 ,r3 ,0x01c
  70018193        | addi       r3 ,r3 ,0x700
  01300093        | addi       r1 ,r0 ,0x013
  0811b8af  	  | amoswap.w  r17,r3 ,r1         
  0000100f        | fence.i 
  6340006f        | jal        r0 ,0x00634
  00000013        | nop
```
First few instructions initialize registers and set interrupt handler base address.
Last 4 instructions do what I said. 

I'm running bare metal programs and I manually set reset vector to 0x80000000 for test.
The code atomically writes value in r1 to 0x80000700 and jump to 0x80000700 after fence.i.

Could anyone check it?

**Please tell us about your environment:**

using freechips.rocketchip.system.DefaultConfig.v

**What is the use case for changing the behavior?**

aswaterman, 2020-03-06:
I tested your code sequence and it worked as expected: when the processor jumps to 0x80000700, it fetches the instruction 0x00000013, which is what the amoswap.w instruction wrote to that location.

JaewonHur, 2020-03-06:
I think there is a bug in my testbench. 

For now, I'm simulating with RocketTile and I wrote testbench for bus slave which follows TileLink spec. 
When I saw the waveform of tl_a ports, RocketTile first Acquire Block of address 0x80000700, then It request GET of that address when it jumps. 

Could you give me the waveform of trace (vcd file)?

Thanks.

aswaterman, 2020-03-06:
I didn't generate one and I already deleted the program.  Just use the
standard testbench in rocket-chip/emulator.  I didn't do anything special;
I just ran

./emulator-freechips.rocketchip.system-DefaultConfig +verbose a.out  2> log

JaewonHur, 2020-03-08:
Thanks, 
There's a bug in my testbench.

aswaterman, 2020-03-09:
Ok, glad you figured it out.

closed 2020-03-09 by aswaterman