chipsalliance/rocket-chip #2345: Mismatch between Rocket chip and Spike

JaewonHur, 2020-03-19:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: proposal

**Other information**
 

Hello,
Recently I'm verifying Rocket chip by cross-checking it with Spike.

There is a mismatch between Spike and Rocket chip.
It seems Rocket chip counts 'ebreak' instruction as returned instruction, while spike doesn't.
Detailed explanation is as below.

**Rocket chip commit number**: 9a1da4e01e3e767451d9ee4a4f0ee8a55a3313da
**Configuration: MemPortOnlyConfig**

When I read **minstret csr**  after **ebreak** instruction, 
Rocket chip and Spike returns different number.

Here's my test executable file, assembly file and all logs.
You can make executable file by typing 'make'.
commands are as follows.

**Rocket chip** 
``` ./emulator-freechips.rocketchip.system-MemPortOnlyConfig-debug +max-cycles=200000 +verbose -v output.vcd test 2>&1 | spike-dasm > test_rocket.out ``` 
**Spike** 
``` spike -l test 2> test_spike.log ``` 

Rocket chip passes but Spike fails for the same test.
Could you please check this behavior?

Thanks.

[test.zip](https://github.com/chipsalliance/rocket-chip/files/4352320/test.zip)

aswaterman, 2020-03-19:
Yeah, Spike is right; technically, ebreak doesn't retire.

In practice, there will be other divergences in instructions-retired count, because of things like LR/SC retry.  So I'm not sure it's worth fixing.

JaewonHur, 2020-03-20:
Thank you for early reply.
If you don't have any plan to fix this problem, may I close this issue?

aswaterman, 2020-03-20:
Yeah, feel free.  If we happen to fix this issue in the future, we'll dig up this issue to reference it.

Thanks for your help!

closed 2020-03-20 by JaewonHur