chipsalliance/rocket-chip #2348: mtval not written after ebreak

JaewonHur, 2020-03-20:
**Type of issue**: bug report 

**Impact**:  unknown

**Development Phase**: proposal

**Other information**
 

Spec says `mtval` register is written with the faulting virtual address when a hardware breakpoint is triggered, but Rocket chip does not save the address to `mtval` register when `ebreak` instruction is executed. Is it intended?

Here's all my test and logs as before.

[test.zip](https://github.com/chipsalliance/rocket-chip/files/4358242/test.zip)

aswaterman, 2020-03-20:
Reading the spec again, I think this might be a Spike bug.  The relevant passage is:

"When a hardware breakpoint is triggered, or an address-misaligned, access-fault, or page fault exception occurs on an instruction fetch, load, or store, mtval is written with the faulting virtual address.  On an illegal instruction trap, [...].  For other traps, mtval is set to zero, but a future standard may redefine mtval's setting for other traps."

`ebreak` is none of those things.  (It's not a hardware breakpoint; that's a different thing.)  So writing 0 to `mtval` is correct operation.

JaewonHur, 2020-03-20:
Thanks for reply.

I will file an issue to spike for further tracking.

devtyagi3909, 2026-09-07:
This was fixed in #3193 (merged Jan 2023) which correctly populates `tval` with the faulting PC when `insn_break` is asserted in `CSR.scala`. The `c.ebreak` path via `RVCExpander` also correctly expands to `0x00100073` and follows the same `tval` datapath. Closing as resolved.